iic-jku/IIC-OSIC-TOOLS #6: GDS files empty in magic when edited with klayout

w32agobot, 2023-02-04:
**Describe the bug**
GDS files can not be viewed with magic if they were edited with klayout

**To Reproduce**
Preparation:
* Start container
* Open terminal in /foss/designs
* `cp -r /foss/tools/openlane/2023.01/designs/spm /foss/designs/spm`
* `flow.tcl -design spm -tag foo`
* `cd spm/runs/foo/results/final/gds/`

Check if gds can be opened with magic
* `magic spm.gds`
* Everything's okay, close magic

Force issue:
* `ke spm.gds`
* Warning: Initialization of layers failed: Unable to open file: /headless/.klayout/tech/sky130/sky130.lyp (errno=2)
* `File` - `Save`
* close klayout

Check again if gds can be opened with magic
* `magic spm.gds`
* magic view is empty

**Expected behavior**
Magic can open the modified gds file

**Environment:**
 - OS: Windows
 - Operating mode: VNC
 - Version tag 2023.01

**Additional context**
 - My workaround is using version tag 2022.12
 - Also happens in tag `next` with `2023.02`

hpretl, 2023-02-05:
This might be related to this issue, to be further investigated: https://github.com/KLayout/klayout/issues/1279

w32agobot, 2023-02-05:
I tiried to workaround by:
copy `/headless/.klayout/tech/sky130` in `2022.12`
paste `/headless/.klayout/tech/sky130` in `latest`

The warning is gone but magic still can't display the modified gds file. 

In `2022.12` the folder `/headless/.klayout/tech/sky130` with `sky130.lyp` exists, but in `latest` it does not. I suspect there's information about the `sky130` path embedded in the openLane-generated gds file (see `file - layout properties` and `tools - manage technologies`), therefore klayout still searches for `sky130.lyp` despite the command line option

hpretl, 2023-02-05:
I think I have found the issue, will try a patch of the klayout techfiles.

hpretl, 2023-02-10:
Closed, for explanation see here https://github.com/RTimothyEdwards/magic/issues/221

closed 2023-02-10 by hpretl