iic-jku/IIC-OSIC-TOOLS #9: gf180mcuC Xschem Setup Bugs

curtisma, 2023-04-23:
**Describe the bug**
We're seeing issues with running Xschem with the gf180mcuC process.  It doesn't seem to be working correctly.  It doesn't have the models or PDK setup correctly within xschem.  The $XSCHEM_LIBRARY_PATH is also not setup correctly.  My teammates are also reporting simulation issues.

I start by sourcing the following `.designinit` file in a fresh terminal.

``` bash
#!/bin/bash
export PDK=gf180mcuC
export PDKPATH=$PDK_ROOT/$PDK
export STD_CELL_LIBRARY=gf180mcu_fd_sc_mcu7t5v0
```

Then I start xschem, but it gives me the following incorrect pdk information in the terminal and it can't find the default process schematic as shown in the following error message pop-up.

``` bash
pdk installation: using /foss/pdks
180MCU_MODELS: /foss/pdks/ngspice/models
```

[image]

This shows that the PDK models and the $XSCHEM_LIBRARY_PATH is not setup correctly.

Doing a diff of the xschemrc from SKY130 and gf180mcuC shows the following as the most significant difference.  I also know several other differences before that which were mostly commented out values but seemed to show the gf180mcuC xschemrc is based off an older version of the xschemrc compared to SKY130.

[image]

I'm going to continue debug by updating to the latest version of the container.  

**To Reproduce**
Steps to reproduce the behavior:
1. Source the .designinit for gf180mcuC as given above
2. run `xschem` command
4. See the errors listed above

**Expected behavior**
I would expect the correct xschem "starter" schematic for the gf180mcuC process to open.
I would also expect the 

**Screenshots**
See above

**Environment:**
 - OS: Windows running the container using Docker desktop which uses the WSL2 kernel
 - Operating mode: X11 using the X410 X-windows server.
 - Version tag: Not sure.  I am using a devcontainer based on some version of it.  
   Is there a way to check the version number from within it?
   I tried checking within Docker and it just shows the layers it used from iic-osic-tools and not the version of the image those came from.

**Additional context**
Add any other context about the problem here.

curtisma, 2023-04-23:
Ok, so I had to make some updates to the gf180mcuC process's xschemrc to get it working.  Mostly I just had to update it to match the sky130 with only a few differences required for gf180mcuC.
The gf180mcuC process's xschemrc is defined at: /foss/pdks/gf180mcuC/libs.tech/xschem/xschemrc

Line 33-34 Original:
``` tcl
append XSCHEM_LIBRARY_PATH :$env(PWD)
# append XSCHEM_LIBRARY_PATH :/mnt/sda7/home/schippes/pdks/sky130A/libs.tech/xschem
```

Line 33-34 Fixed:
``` tcl
append XSCHEM_LIBRARY_PATH :[file dirname [info script]]
```

Line 57 original:
``` tcl
set XSCHEM_START_WINDOW {gf180mcutests/0_top.sch}
```

Line 57 Fixed:
``` tcl
set XSCHEM_START_WINDOW {tests/0_top.sch}
```
This will now work since the /foss/pdks/gf180mcuC/libs.tech/xschem directory is now on $XSCHEM_LIBRARY_PATH because of the line 33 fix.

I then updated the last section of the file to the following:

``` tcl
###########################################################################
#### PDK SPECIFIC VARIABLES
###########################################################################

## check if env var PDK_ROOT exists, and use it for building open_pdks paths
if { [info exists env(PDK_ROOT)] && $env(PDK_ROOT) ne {} } {
  ## found variable, set tcl PDK_ROOT var
  if {![file isdir $env(PDK_ROOT)]} {
    puts stderr "Warning: PDK_ROOT environment variable is set but path not found on the system."
  }
  set PDK_ROOT $env(PDK_ROOT)
} else {
  ## not existing or empty. 
  puts stderr "Warning: PDK_ROOT env. var. not found or empty, trying to find an open_pdks install"
  if {[file isdir /usr/share/pdk]} {set PDK_ROOT /usr/share/pdk
  } elseif {[file isdir /usr/local/share/pdk]} {set PDK_ROOT /usr/local/share/pdk
  } elseif {[file isdir $env(HOME)/share/pdk]} {set PDK_ROOT $env(HOME)/share/pdk
  } else {
    puts stderr {No open_pdks installation found, set PDK_ROOT env. var. and restart xschem}
  }
}

if {[info exists PDK_ROOT]} {
  ## get process variant
  if {[info exists env(PDK)]} {
    set PDK $env(PDK)
  } else {
    set PDK gf180mcuC
  }
  set 180MCU_MODELS ${PDK_ROOT}/${PDK}/libs.tech/ngspice/models
  puts stderr "pdk installation: using ${PDK_ROOT}/${PDK}"
  puts stderr "180MCU_MODELS: $180MCU_MODELS"
  append XSCHEM_LIBRARY_PATH :${PDK_ROOT}/${PDK}/libs.tech/xschem
  puts stderr "XSCHEM_LIBRARY_PATH: \n  $XSCHEM_LIBRARY_PATH"
}
```

This shows the main changes in that last section compared to sky130A xschemrc:

[image]

curtisma, 2023-04-23:
Does this change also need to be made in the openpdks repo?

I'm also not seeing the update to append $XSCHEM_USER_LIBRARY_PATH to ${XSCHEM_LIBRARY_PATH}.
I see that a new image was released yesterday.  Do you need to update the open_pdk version number in it?

hpretl, 2023-04-26:
I think this should be addressed in `open_pdks`, I have created this issue over there: https://github.com/RTimothyEdwards/open_pdks/issues/368

hpretl, 2023-04-26:
BTW, I ran the same procedure you describe above, and I get this error window, but no printouts in the starting shell.

[image]

curtisma, 2023-04-29:
hmm, in order to get the output in the shell, the "-b" option needs to be removed from the "~/.bashrc" or to use the xschemtcl command.  I would make the default to be without "-b":

Remove it from line 118 and remove the duplicate alias on line 119:

```diff
- alias xschem='xschem -b --rcfile $PDKPATH/libs.tech/xschem/xschemrc'
+alias xschem='xschem --rcfile $PDKPATH/libs.tech/xschem/xschemrc'
- alias xschemtcl='xschem --rcfile $PDKPATH/libs.tech/xschem/xschemrc'

```

"-b" option definition:
``` bash
usage: xschem [options] [schematic | symbol ]
Options:
  -h  --help           Print this help.
  -b  --detach         Detach Xschem from console (no output and no input from console)
  ...
```

curtisma, 2023-04-29:
@hpretl Your error message is indicating the update on line 57 wasn't made.

Line 57 update:
```diff
- set XSCHEM_START_WINDOW {gf180mcutests/0_top.sch}
+ set XSCHEM_START_WINDOW {tests/0_top.sch}
```

The update on line 340 of the new version adds the parent path of the "tests" folder to the `$XSCHEM_LIBRARY_PATH`.

``` bash
append XSCHEM_LIBRARY_PATH :${PDK_ROOT}/${PDK}/libs.tech/xschem
```

curtisma, 2023-04-29:
I found an error with the model path that I had in the original.  the model path 

Update on line 339 near the end of the new 
```diff

if {[info exists PDK_ROOT]} {
  ## get process variant
  if {[info exists env(PDK)]} {
    set PDK $env(PDK)
  } else {
    set PDK gf180mcuC
  }
-  set 180MCU_MODELS ${PDK_ROOT}/${PDK}/libs.tech/ngspice/models
+  set 180MCU_MODELS ${PDK_ROOT}/${PDK}/libs.tech/ngspice
  puts stderr "pdk installation: using ${PDK_ROOT}/${PDK}"
  puts stderr "180MCU_MODELS: $180MCU_MODELS"
  append XSCHEM_LIBRARY_PATH :${PDK_ROOT}/${PDK}/libs.tech/xschem
  puts stderr "XSCHEM_LIBRARY_PATH: \n  $XSCHEM_LIBRARY_PATH"
}
```

Now the simulations are working:
[image]

curtisma, 2023-04-29:
I'll update the fix info in the first comment of this issue and provide the file I've been using so that the full file can be overwritten rather than having to do line edits.

closed 2023-07-11 by hpretl