iic-jku/IIC-OSIC-TOOLS #13: Create an "iic-osic-tools-analog" image with smaller size

curtisma, 2023-07-15:
##Is your feature request related to a problem? Please describe.
TL;DR:  For analog-only designs, the large image size is using a lot more GitHub codespaces credits than necessary for digital design tools that are unused. 

The size of the full image with both the analog and digital design tools takes around 40GB of size when running. (the compresses image is ~5GB and the image on disk ~20GB. 

I've been using the image as part of a devcontainer that is used by Github codespaces.  A design project repo can then be started in a complete analog design environment in the cloud with just a couple of clicks from the GitHub repo webpage.
However the size of this image is causing me to have to use a codespace with a lot of hardware resources.  (currently using the largest with 16 cores / 32GB ram / 128GB disk space).  GitHub limits the number of codespace credits based on the hardware size used.  So it will end up costing me more for a bunch of digital tools that I don't use in a pure analog design.

##Describe the solution you'd like

I'd like to have a second smaller image, "iic-osic-tools-analog" with only the tools necessary for analog design. 

Hopefully the build wouldn't take much longer since the analog tool layers could be cached and reused between the 2 images. 
 The builds of the common tools can be cached in Docker image layers so they can be reused between both images.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.
- Create a separate repo for the analog image.  But this would make it harder to maintain and keep in sync with updated to the full image.  It also wouldn't take advantage of the caching between the 2 images.

**Additional context**
Add any other context or screenshots about the feature request here.

hpretl, 2023-09-05:
We are internally discussing this for some time already; so far, we have shied the effort. A large piece of the image size is the PDKs, which would be needed in any case. Arguably, the largest piece of the PDKs are the digital standard cells, which could be removed (on one hand), but are sometimes used in analog/mixed, so should be kept (on the other hand).

A distinction between an "analog" and a "full-featured" image is not so clear-cut in my view, and I fear a slow feature creep on a minimal analog image over time by different people asking for different components, ending up in almost full size in foreseeable future :-)

Having said that, the (growing) image size is a concern. What the right answer is, I don't know yet.

If you have a clear picture of which components you would like to see in an analog image, please let us know.

hpretl, 2025-06-13:
Potentially, we can support this request after revamping our build process, see https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/138

MrHighVoltage, 2025-08-26:
This is now doable with the new build-system, by creating separate Dockerfiles for the final build stage.  By only a copying a reduced amount of packages to the final image. See https://github.com/iic-jku/IIC-OSIC-TOOLS/blob/main/_build/images/iic-osic-tools/Dockerfile

hpretl, 2026-10-27:
With the build rework from #138, an analog image is now a matter of a reduced final Dockerfile plus a bake target. The work is tracked in #398, together with the digital and RISC-V flavors. Closing this in favor of #398.

closed 2026-10-27 by hpretl