iic-jku/IIC-OSIC-TOOLS #14: Current repository configuration `main` branch not building

daquintero, 2023-08-23:
**Describe the bug**
I've just cloned the repo and tried to build it in my local Docker to make some edits, and the latest `main` branch is not building.

**To Reproduce**
Steps to reproduce the behavior:
```
11:10:26 dq16394@iqt iic-osic-tools ±|main|→ ls
build-all.sh      Dockerfile             eda_server_start.sh  LICENSE           start_jupyter.bat  start_vnc.bat  start_x.sh             tools
builder-clear.sh  eda_server_conf.sh     eda_server_stop.sh   README.md         start_jupyter.sh   start_vnc.sh   tool_metadata_add.yml
buildkitd.toml    eda_server_restart.sh  images               RELEASE_NOTES.md  start_shell.sh     start_x.bat    tool_metadata.yml
11:10:37 dq16394@iqt iic-osic-tools ±|main|→ docker build . -t daquintero/iic
[+] Building 1690.5s (57/129)                                                                                                                                                 docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                    0.1s
 => => transferring dockerfile: 20.03kB                                                                                                                                                 0.0s
 => [internal] load .dockerignore                                                                                                                                                       0.1s
 => => transferring context: 2B                                                                                                                                                         0.0s
 => [internal] load metadata for docker.io/library/ubuntu:jammy                                                                                                                         1.6s
 => [internal] load build context                                                                                                                                                       0.9s
 => => transferring context: 27.81MB                                                                                                                                                    0.7s
 => [base 1/3] FROM docker.io/library/ubuntu:jammy@sha256:ec050c32e4a6085b423d36ecd025c0d3ff00c38ab93a3d71a460ff1c44fa6d77                                                              4.9s
 => => resolve docker.io/library/ubuntu:jammy@sha256:ec050c32e4a6085b423d36ecd025c0d3ff00c38ab93a3d71a460ff1c44fa6d77                                                                   0.0s
 => => sha256:56887c5194fddd8db7e36ced1c16b3569d89f74c801dc8a5adbf48236fb34564 424B / 424B                                                                                              0.0s
 => => sha256:01f29b872827fa6f9aed0ea0b2ede53aea4ad9d66c7920e81a8db6d1fd9ab7f9 2.30kB / 2.30kB                                                                                          0.0s
 => => sha256:b237fe92c4173e4dfb3ba82e76e5fed4b16186a6161e07af15814cb40eb9069d 29.54MB / 29.54MB                                                                                        1.0s
 => => sha256:ec050c32e4a6085b423d36ecd025c0d3ff00c38ab93a3d71a460ff1c44fa6d77 1.13kB / 1.13kB                                                                                          0.0s
 => => extracting sha256:b237fe92c4173e4dfb3ba82e76e5fed4b16186a6161e07af15814cb40eb9069d                                                                                               3.5s
 => [base 2/3] COPY images/base/scripts/dependencies.sh dependencies.sh                                                                                                                 4.1s
 => [base 3/3] RUN bash dependencies.sh                                                                                                                                              1469.5s
 => [padring 1/2] COPY images/padring/scripts/install.sh install.sh                                                                                                                     0.3s
 => [align-pdk-sky130 1/2] COPY images/align-pdk-sky130/scripts/install.sh install.sh                                                                                                   0.6s
 => [yosys 1/2] COPY images/yosys/scripts/install.sh install.sh                                                                                                                         0.3s
 => [xyce 1/4] COPY images/xyce/scripts/trilinos.reconfigure.sh /trilinos.reconfigure.sh                                                                                                0.5s
 => [nvc 1/2] COPY images/nvc/scripts/install.sh install.sh                                                                                                                             0.6s
 => [openroad_app 1/2] COPY images/openroad/scripts/install.sh install.sh                                                                                                               0.5s
 => [gtkwave 1/2] COPY images/gtkwave/scripts/install.sh install.sh                                                                                                                     0.7s
 => [covered 1/2] COPY images/covered/scripts/install.sh install.sh                                                                                                                     1.5s
 => [gaw3-xschem 1/2] COPY images/gaw3-xschem/scripts/install.sh install.sh                                                                                                             0.8s
 => [xschem 1/2] COPY images/xschem/scripts/install.sh install.sh                                                                                                                       0.9s
 => [rftoolkit 1/2] COPY images/rftoolkit/scripts/install.sh install.sh                                                                                                                 0.9s
 => [ngspice 1/2] COPY images/ngspice/scripts/install.sh install.sh                                                                                                                     1.2s
 => [iverilog 1/2] COPY images/iverilog/scripts/install.sh install.sh                                                                                                                   1.0s
 => [netgen 1/2] COPY images/netgen/scripts/install.sh install.sh                                                                                                                       0.7s
 => [riscv-gnu-toolchain-rv32i 1/2] COPY images/riscv-gnu-toolchain-rv32i/scripts/install.sh install.sh                                                                                 1.0s
 => [qucs-s 1/2] COPY images/qucs-s/scripts/install.sh install.sh                                                                                                                       1.3s
 => [fault 1/4] COPY images/fault/scripts/dependencies.sh dependencies.sh                                                                                                               1.5s
 => [irsim 1/2] COPY images/irsim/scripts/install.sh install.sh                                                                                                                         1.3s
 => [verilator 1/2] COPY images/verilator/scripts/install.sh install.sh                                                                                                                 1.3s
 => [qflow 1/2] COPY images/qflow/scripts/install.sh install.sh                                                                                                                         1.1s
 => [cvc_rv 1/2] COPY images/cvc_rv/scripts/install.sh install.sh                                                                                                                       1.1s
 => [basepkg 1/2] COPY images/base/scripts/install.sh install.sh                                                                                                                        1.1s
 => [slang 1/2] COPY images/slang/scripts/install.sh install.sh                                                                                                                         1.4s
 => [ghdl 1/2] COPY images/ghdl/scripts/install.sh install.sh                                                                                                                           1.4s
[... cut ...]
 178 |     COPY images/ngspice/scripts/install.sh install.sh
 179 | >>> RUN bash install.sh
 180 |
 181 |     #######################################################################
--------------------
ERROR: failed to solve: process "/bin/sh -c bash install.sh" did not complete successfully: exit code: 127

```

**Expected behavior**
The environment to build.

**Environment:**
Docker in a `CentOS 7` environment that does have the OpenLane docker working.

**Additional context**
I plan to try to solve this but just flagging this in the interim.

daquintero, 2023-08-23:
Running the `build-all` script makes the terminal crash.

daquintero, 2023-08-23:
Differnet issue now:

```
12:13:16 dq16394@iqt ~ ±|master|→ cd iic-osic-tools/
/home/dq16394/iic-osic-tools
12:13:19 dq16394@iqt iic-osic-tools ±|main|→ ls
build-all.sh           images             start_vnc.sh
builder-clear.sh       LICENSE            start_x.bat
buildkitd.toml         README.md          start_x.sh
Dockerfile             RELEASE_NOTES.md   tool_metadata_add.yml
eda_server_conf.sh     start_jupyter.bat  tool_metadata.yml
eda_server_restart.sh  start_jupyter.sh   tools
eda_server_start.sh    start_shell.sh
eda_server_stop.sh     start_vnc.bat
12:13:20 dq16394@iqt iic-osic-tools ±|main|→ docker build . -t daquintero/iic
[+] Building 601.3s (59/129)                                     docker:default> [slang 2/2] RUN bash install.sh                                      598.
 => [slang 2/2] RUN bash install.sh                                      598.9s> [riscv-gnu-toolchain-rv32i 2/2] RUN bash install.sh                  598.
 => [qucs-s 2/2] RUN bash install.sh                                     598.9s> [ngspice 2/2] RUN bash install.sh                                    598.
 => [riscv-gnu-toolchain-rv32i 2/2] RUN bash install.sh                  598.9s> [xyce 4/4] RUN bash install.sh                                       598.
 => [verilator 2/2] RUN bash install.sh                                  598.9s> => # es/teuchosparameterlist.dir/Teuchos_XMLParameterListWriter.cpp.o
 => [ngspice 2/2] RUN bash install.sh                                    598.9s> => # es/teuchosparameterlist.dir/Teuchos_XMLParser.cpp.o
 => [openroad_app 2/2] RUN bash install.sh                               598.9s> => # es/teuchosparameterlist.dir/Teuchos_XMLPerfTestArchive.cpp.o
 => [xyce 4/4] RUN bash install.sh                                       598.9s> [ngspyce 1/2] COPY images/ngspyce/scripts/install.sh install.sh        0.
 => => # [ 29%] Building CXX object packages/teuchos/parameterlist/src/CMakeFil> [pyopus 2/2] RUN bash install.sh                                      58.
 => => # es/teuchosparameterlist.dir/Teuchos_XMLParameterListWriter.cpp.o      > [align 2/2] RUN bash install.sh                                       58.
 => => # [ 29%] Building CXX object packages/teuchos/parameterlist/src/CMakeFil> [ngspyce 2/2] RUN bash install.sh                                     24.
 => => # es/teuchosparameterlist.dir/Teuchos_XMLParser.cpp.o                 [+] Building 601.4s (59/129)                                     docker:default> => # [ 29%] Building CXX object packages/teuchos/parameterlist/src/CMakeF => [slang 2/2] RUN bash install.sh                                      599.1s> => # es/teuchosparameterlist.dir/Teuchos_XMLPerfTestArchive.cpp.o         => [qucs-s 2/2] RUN bash install.sh                                     599.1s> [magic 1/2] COPY images/magic/scripts/install.sh install.sh            0. => [riscv-gnu-toolchain-rv32i 2/2] RUN bash install.sh                  599.1s> [ngspyce 1/2] COPY images/ngspyce/scripts/install.sh install.sh        0. => [verilator 2/2] RUN bash install.sh                                  599.1s> [klayout 1/2] COPY images/klayout/scripts/install.sh install.sh        0. => [ngspice 2/2] RUN bash install.sh                                    599.1s> [pyopus 2/2] RUN bash install.sh                                      59. => [openroad_app 2/2] RUN bash install.sh                               599.1s> [openlane 2/2] RUN bash install.sh                                     9. => [xyce 4/4] RUN bash install.sh                                       599.1s> [align 2/2] RUN bash install.sh                                       59. => => # [ 29%] Building CXX object packages/teuchos/parameterlist/src/CMakeFil> [klayout 2/2] RUN bash install.sh                                     58. => => # es/teuchosparameterlist.dir/Teuchos_XMLParameterListWriter.cpp.o      > [ngspyce 2/2] RUN bash install.sh                                     24. => => # [ 29%] Building CXX object packages/teuchos/parameterlist/src/CMakeFil> [magic 2/2] RUN bash install.sh                                       58. => => # es/teuchosparameterlist.dir/Teuchos_XMLParser.cpp.o                 [+] Building 601.6s (59/129)                                     docker:default> => # [ 29%] Building CXX object packages/teuchos/parameterlist/src/CMakeF => [slang 2/2] RUN bash install.sh                                      599.2s> => # es/teuchosparameterlist.dir/Teuchos_XMLPerfTestArchive.cpp.o         => [qucs-s 2/2] RUN bash install.sh                                     599.2s> [magic 1/2] COPY images/magic/scripts/install.sh install.sh            0. => [riscv-gnu-toolchain-rv32i 2/2] RUN bash install.sh                  599.2s> [ngspyce 1/2] COPY images/ngspyce/scripts/install.sh install.sh        0. => [verilator 2/2] RUN bash install.sh                                  599.2s> [klayout 1/2] COPY images/klayout/scripts/install.sh install.sh        0. => [ngspice 2/2] RUN bash install.sh                                    599.2s> [pyopus 2/2] RUN bash install.sh                                      59. => [openroad_app 2/2] RUN bash install.sh                               599.2s> [openlane 2/2] RUN bash install.sh                                     9. => [xyce 4/4] RUN bash install.sh                                       599.2s> [align 2/2] RUN bash install.sh                                       59. => => # [ 29%] Building CXX object packages/teuchos/parameterlist/src/CMakeFil> [klayout 2/2] RUN ba [... cut ...]
862.1 [ 74%] Linking CXX static library libcomponents.a
892.0 [ 74%] Built target components
898.2 make: *** [Makefile:136: all] Error 2
------
Dockerfile:259
--------------------
 257 |     ARG QUCS_NAME="qucs-s"
 258 |     COPY images/qucs-s/scripts/install.sh install.sh
 259 | >>> RUN bash install.sh
 260 |
 261 |     #######################################################################
--------------------
ERROR: failed to solve: process "/bin/sh -c bash install.sh" did not complete successfully: exit code: 2
```

daquintero, 2023-08-23:
I uncommented it out temporarily and also OpenRoad is not building. I think I now fully appreciate how complicated it is to mantain this environment!

```
1245.2 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb2Core.c.o
1246.4 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb2Driver.c.o
1247.7 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb2Dump.c.o
1248.0 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb2Flow.c.o
1249.1 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb2Image.c.o
1249.3 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb3Image.c.o
1249.4 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb3Nonlin.c.o
1250.0 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Cex.c.o
1250.0 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Image.c.o
1250.0 [ 64%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Nonlin.c.o
1250.4 [ 65%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/bdd/llb/llb4Sweep.c.o
1259.8 [ 65%] Linking CXX static library libabc.a
1264.5 [ 65%] Built target libabc
1264.6 make: *** [Makefile:146: all] Error 2
------
Dockerfile:219
--------------------
 217 |     ARG OPENROAD_APP_NAME="openroad"
 218 |     COPY images/openroad/scripts/install.sh install.sh
 219 | >>> RUN bash install.sh
 220 |
 221 |     #######################################################################
--------------------
ERROR: failed to solve: process "/bin/sh -c bash install.sh" did not complete successfully: exit code: 2
```

hpretl, 2023-08-23:
I have recently (last week) built the mainline and it was working fine, so it is a bit tough to debug what is going wrong on your end. As you comment, build this large amount of packages is a pain :-) What are you using for building? We build the `amd64` version on an Ubuntu machine, and the `arm64` version on a macOS (MacBook)?

hpretl, 2023-09-17:
I merged PR#6, but had to remove `jaxlib` (not available for `arm64`) and `piel` (I think the `femwell` dependency is not available).

closed 2023-12-02 by hpretl