iic-jku/IIC-OSIC-TOOLS #16: find in the start_x.bat cannot work

ll550, 2023-09-21:
**Describe the bug**
when run the start_x.bat with a cotainner running, the find function is not working in the command prompt.

docker container inspect %CONTAINER_NAME% 2>&1 | find "Status" | find /i "running"
find: '/i': No such file or directory
find:find :'Status' : 'running'No such file or directory:
No such file or directory

Replacing the find by findstr, everything is OK

**To Reproduce**
Let the docker image running, type start_x.bat in the command prompt.

**Expected behavior**
The batch file is supposed to find the running status correctly.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment:**
 - OS: Win10
 - Operating mode: X11
 - Version tag: latest

**Additional context**
Add any other context about the problem here.

hpretl, 2023-09-21:
@MrHighVoltage Can you please take a look and fix? Thx!

MrHighVoltage, 2023-09-22:
Hi, yes I will take a look. Very interesting that this doen't work anymore, because I probably tested this.
Thank you for the Issue, I'm looking at it and keep you updated.

MrHighVoltage, 2023-09-22:
So I just tested it on a Windows machine and looked into find vs. findstr and it seems that findstr just introduces Regex to find. @ll550, could you maybe give me more information, which shell are you using etc.?

ll550, 2023-09-22:
@MrHighVoltage hi 
I just used cmd.exe, which is offered by the os win10.

MrHighVoltage, 2023-09-22:
That is very interesting, because it works on two different machines for me. @ll550 Do you maybe have some kind of Unix-Environment installed? Like cygwin? It could happen that it overwrites the "windows find" with the "cygwin find". You can checkt that with "which find".

Anyways, I'll change it to findstr, which seems to be more robust if something like that happens, because there is no findstr in the unix world, afaik.

closed 2023-09-22 by MrHighVoltage

ll550, 2023-09-22:
@MrHighVoltage  Yes, you are right. I used to install msys64 in the computer and leave it for nothing. Sorry to bother.