iic-jku/IIC-OSIC-TOOLS #17: Bash Interface to automatise the tasks in Xschem NGSpice

sras202, 2023-11-20:
Hello,

I am using Xschem with NGSpice at the backend through the Docker station provided by Efabless.com for circuit simulation. I have 100 different trials of the input signal, each saved in a separate text file (.txt). Manually loading timestamps and voltage values for each trial in the piecewise linear function, and modifying the output file path (as it's different for every trial) is time-consuming for 100 trials.

I'm looking to automate this task of loading different input files, simulating, and saving as different output files through a bash script. Has anyone implemented a bash interface to automatically load input data and save it as different output files through a loop? If so, is it possible to implement it inside the Docker container? If not, I'd be happy to contribute in this direction.

Thanks and Regards,
Shavika Rastogi
PhD Candidate
International Centre for Neuromorphic Systems
Western Sydney University, Australia
and
Biocomputation Research Group
University of Hertfordshire, UK

MrHighVoltage, 2023-11-20:
This is already possible with our container, but requires a little bit of manual setup. First, a good starting-point to "generate" a starting command is to use the "DRY_RUN" option of our start scripts. For example you can run `DRY_RUN=1 ./start_shell.sh`, and the shellscript prints out the commands it would actually use to start the container, in my case this looks like:

`docker run -it --name iic-osic-tools_shell_uid_1000 --user 0:0 -e DISPLAY= -v /home/georg/eda/designs:/foss/designs:rw hpretl/iic-osic-tools:latest -s /bin/bash`

For your application you can change a few things in this command line.

- After `-it`, you should add `--rm`, which automatically deletes the container, when your job is done. Just keep an eye on the output files, everything must be in the /foss/designs directory in the container, or it will be deleted.
- You can remove the name parameter, as it is deleted automatically and therefore not really useful.
- The `--user 0:0` should be adapted to fit your current user id and group id. Typically this is 1000:1000 on a local Linux install. You can find it with `id -u` and `id -g`
- The actuall command you want to run is placed after `-s`. In our startup script, this mode is basically "skip startup" and it just runs the given command and exists. As you can see in this mode for start_shell, only a bash is launched and nothing else happens. For your application, you put the script to execute into your hosts design directory. When launching the container, the final command line would be something like:

`docker run -it --rm --name batch-runner-container --user 1000:1000 -v /path-to-host-home/eda/designs:/foss/designs:rw hpretl/iic-osic-tools:latest -s /bin/bash /foss/designs/my_runner.sh my_runner_parameters`

Thank you for the input, we will add a new start script with the comming release!

sras202, 2023-11-21:
Hello, 

I did the required DRY RUN and ran the final command which looked something like this in my case:

docker run -it --rm --name batch-runner-container --user 1000:1000 -v /home/shavika/eda/designs:/foss/designs:rw hpretl/iic-osic-tools:latest -s /bin/bash/foss/designs/my_runner.sh my_runner_parameters

But it couldn't find the directory : 

[INFO] USER_ID: 1000, GROUP_ID: 1000
[INFO] Final PATH variable: /foss/tools/bin:/foss/tools/sak/:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/foss/tools/covered/19d30fc/bin:/foss/tools/cvc_rv/df85a63/bin:/foss/tools/fault/54e8665:/foss/tools/gaw3-xschem/a4bb956/bin:/foss/tools/gds3d/173da0c/bin:/foss/tools/ghdl/v3.0.0/bin:/foss/tools/gtkwave/8e0bcdb/bin:/foss/tools/irsim/16777a6/bin:/foss/tools/iverilog/09f3ebf/bin:/foss/tools/klayout/44a2aa9:/foss/tools/magic/0afe4d8/bin:/foss/tools/netgen/87d8759/bin:/foss/tools/ngspice/ngspice/bin:/foss/tools/nvc/r1.10.1/bin:/foss/tools/openlane/2023.08:/foss/tools/openroad/0cfb9a4/bin:/foss/tools/osic-multitool:/foss/tools/padring/b2a64ab/bin:/foss/tools/pyopus/0.11/bin:/foss/tools/qflow/b0f76bf/bin:/foss/tools/qucs-s/1.1.0/bin:/foss/tools/rftoolkit/bin:/foss/tools/slang/4decdb5/bin:/foss/tools/verilator/947402b/bin:/foss/tools/xschem/e3b9e67/bin:/foss/tools/xyce/Parallel/Release/bin:/foss/tools/yosys/14d50a1/bin:/foss/tools/align-utils
[INFO] Final PYTHONPATH variable: /usr/lib/python310.zip:/usr/lib/python3.10:/usr/lib/python3.10/lib-dynload:/usr/local/lib/python3.10/dist-packages:/usr/lib/python3/dist-packages:/foss/tools/ngspyce/154a272/local/lib/python3.10/dist-packages:/foss/tools/align/d3954af/local/lib/python3.10/dist-packages:/foss/tools/pyopus/0.11/local/lib/python3.10/dist-packages
[INFO] SKIPPING UI STARTUP
[INFO] Executing command: '/bin/bash/foss/designs/my_runner.sh my_runner_parameters'
/dockerstartup/scripts/ui_startup.sh: line 38: /bin/bash/foss/designs/my_runner.sh: Not a directory

Is there a way to extract the correct location of the shell script my_runner.sh ?

Thanks in advance.

MrHighVoltage, 2023-11-28:
Sorry, this was a bit of a missunderstanding here. The script name "my_runner.sh" was just a placeholder for something you could use to call the simulator, or whatever jobs you want to run.

With what I proposed, you copy your data for the simulations to the directory that will be mounted. With the command, the container gets spun up, the script will be run, and after it is done, all results in the mounted directory are kept, and everything else is thrown away and does not eat up storage.

For your specific command, the my_runner.sh does not exist, but also, you are missing a space between /bin/bash and the path to the actual script.

closed 2023-12-02 by hpretl