iic-jku/IIC-OSIC-TOOLS #24: Running the Octagon OpenEMS example from IHP fails with ImportError

DanielGerlicher, 2023-12-01:
**Describe the bug**
When trying to execute the example below, there is a python3 ImportError...

```
/foss/pdks/ihp-sg13g2/libs.tech/openems/testcase/SG13_Octagon_L2n0/OpenEMS_Python > python3 SG13_L2n0_mesh1um_v2.py

Traceback (most recent call last):
  File "/foss/pdks/ihp-sg13g2/libs.tech/openems/testcase/SG13_Octagon_L2n0/OpenEMS_Python/SG13_L2n0_mesh1um_v2.py", line 16, in <module>
    from pylab import *
  File "/usr/local/lib/python3.10/dist-packages/pylab.py", line 1, in <module>
    from matplotlib.pylab import *
  File "/usr/local/lib/python3.10/dist-packages/matplotlib/pylab.py", line 44, in <module>
    from matplotlib import cbook, mlab, pyplot as plt
  File "/usr/local/lib/python3.10/dist-packages/matplotlib/pyplot.py", line 66, in <module>
    from matplotlib.figure import Figure, FigureBase, figaspect
  File "/usr/local/lib/python3.10/dist-packages/matplotlib/figure.py", line 43, in <module>
    from matplotlib import _blocking_input, backend_bases, _docstring, projections
  File "/usr/local/lib/python3.10/dist-packages/matplotlib/projections/__init__.py", line 58, in <module>
    from mpl_toolkits.mplot3d import Axes3D
  File "/usr/lib/python3/dist-packages/mpl_toolkits/mplot3d/__init__.py", line 1, in <module>
    from .axes3d import Axes3D
  File "/usr/lib/python3/dist-packages/mpl_toolkits/mplot3d/axes3d.py", line 23, in <module>
    from matplotlib import _api, cbook, docstring, _preprocess_data
ImportError: cannot import name 'docstring' from 'matplotlib' (/usr/local/lib/python3.10/dist-packages/matplotlib/__init__.py)
```

Maybe this is related to https://stackoverflow.com/q/77235006

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '/foss/pdks/ihp-sg13g2/libs.tech/openems/testcase/SG13_Octagon_L2n0/OpenEMS_Python'
2. Run 'python3 SG13_L2n0_mesh1um_v2.py'
3. See error

**Expected behavior**
The testcase should run

**Environment:**
 - OS: Linux
 - Operating mode: VNC
 - Version tag: hpretl/iic-osic-tools:latest (441fc2b128b4)

DanielGerlicher, 2023-12-01:
OpenEMS is installed from APT (including python3-matplotlib and python3-numpy), which seems to collide with the pip packages of matplotlib and numpy.

Would it help switching to a non-apt installation?

MrHighVoltage, 2023-12-01:
Yes, this is a known issue, we are working on a fix. What you can do for now is reinstalling matplotlib from/with pip.
First, open a Terminal on your host and open a root shell on your running container with `docker exec -it -u 0:0 <container name> /bin/bash` and then  `pip install matplotlib --force`.
After this, there is probably another bug coming up with the python interface. You have to edit the "ports.py" file (error shows you the location) and change ".float" to ".float32".

With those steps I could get it to work in my case.

DanielGerlicher, 2023-12-02:
Thanks a lot for the workaround, after one additional step it worked for me!

Here are the steps I followed:

1) Run `docker run --rm -it --user 1000:1000 -p 80:80 -p 5901:5901 hpretl/iic-osic-tools:latest`
2) Execute a root shell using `docker exec -it -u  0:0 582b2e11fc2b /bin/bash`
3) Run `pip install matplotlib --force`
4) Replace "np.float" by "np.float32" in /usr/lib/python3/dist-packages/openEMS/ports.py
5) Edit /foss/pdks/ihp-sg13g2/libs.tech/openems/testcase/SG13_Octagon_L2n0/OpenEMS_Python/SG13_L2n0_mesh1.5um_v2.py in line 533 and hardcode "openEMS" for the binary and remove the import statement
6) Execute the python script

MrHighVoltage, 2024-02-09:
Update on this:
We fixed a few build problems (matplotlib and numpy have been pulled in from apt and with pip, now only from apt). That resulted in Version mismatches, which caused the problem with np.float and "cannot import name 'docstring'". The remaining issues and their workaround are documented in  [KNOWN_ISSUES.md](https://github.com/iic-jku/IIC-OSIC-TOOLS/blob/main/KNOWN_ISSUES.md).

I'll close this issue for now, but we will probably build openEMS from source in the future (most likely when we switch the base OS to the upcoming Ubuntu 24.04 LTS, as quite some reworking will be required anyways).

closed 2024-02-09 by MrHighVoltage