iic-jku/IIC-OSIC-TOOLS #27: define default xschem as "xschem -b" prevent the xschemtcl to enter tcl cmd

cnMuggle, 2024-01-07:
**Describe the bug**
the original xschem cmd from stefan when used, will return a TCL cmd. The default setting in iic-osic is to alias xschem as "xschem -b" in ~/.bashrc file. This setting results in the other alias "xschemtcl", which is supposed to behave like original, output TCL cmd, doesn't work and behave the same as "xschem -b". 

**To Reproduce**
Steps to reproduce the behavior:
1. I use icc-osic-tool-xserver in windows OS
2. after start it, the cmd shows up, type "xschemtcl"
3. the return doesn't have TCL cmd and hence I must have & to send it to the background.
[image]

**Expected behavior**
The below results are obtained via the tricks in the Additional context.
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment:**
 - OS: Windows
 - Operating mode: X11
 - Version tag : **where to find this info? I download from github last month on Dec/2023**

**Additional context**
One workaround is to define "xschem -b" as something else, but not replace the app name "xschem". What I show below in .bashrc is a possible work around. The line of "unalias xschem" is needed for de-link the existing xschem from 'xschem -b'
[image]

It looks like even xschemtcl will call the "xschem -b" if alias xschem is renmaed as 'xschem -b' no matter I put the xschemtcl alias definition before or after the 'xschem -b' alias. I have tried to code in the below way but xschemtcl still doesn't function properly.
[image]

I don't plan to file a Pull request because I think it's kind of personal preference to overwrite 'xschem' as 'xschem -b' and not using xschem_b or something like that. It's just unfortunately that somehow the defined "xschemtcl" doesn't work anymore. A more fundamental solution should fix this issue somewhere else, maybe in BASH, which I am not good at.

hpretl, 2024-01-08:
Thank you, you raised an important point. The solution is rather simple, as using `\` an alias is not used. So we will update this accordingly in the next release:

`alias xschemtcl='\xschem --rcfile $PDKPATH/libs.tech/xschem/xschemrc'`

hpretl, 2024-01-08:
Fixed: https://github.com/iic-jku/IIC-OSIC-TOOLS/commit/726e50996e26d552ea49209da09603568848eb1f

closed 2024-01-08 by hpretl