iic-jku/IIC-OSIC-TOOLS #31: [Image Build Phase] install_ihp.sh can't find openvaf

martinjankoehler, 2024-01-09:
**Describe the bug**
- Tried to build the docker image
- using next_release https://github.com/iic-jku/IIC-OSIC-TOOLS/commit/726e50996e26d552ea49209da09603568848eb1f 

```
[6/39] STEP 7/8: COPY images/open_pdks/scripts/install_ihp.sh install_ihp.sh
--> 9ac5b689a82a
[6/39] STEP 8/8: RUN bash install_ihp.sh 
Cloning into 'ihp'...
Updating files: 100% (3664/3664), done.
Branch 'dev' set up to track remote branch 'dev' from 'origin'.
Switched to a new branch 'dev'
install_ihp.sh: line 40: /foss/tools//openvaf
osic-multitool/bin/openvaf: No such file or directory
Error: building at STEP "RUN bash install_ihp.sh": while running runtime: exit status 127
```

**Expected behavior**
Build to succeed.

hpretl, 2024-01-09:
This is strange. I just built 586314645fd2950b11043b2fda9a086124a0e943 and it built fine.

martinjankoehler, 2024-01-09:
OK, the reason seems to be that `OPENVAF_NAME` is empty,
then the `ls` command will return a multi-line result.

[image]

---

As our administrators do not allow the regular Docker (no `root` rights for users),
we have to use Podman as a drop-in replacement (allowing building images and running containers without root rights).

Perhaps Podman behaves differently regarding making `ARG` available as `ENV` in follow-up stages.

The `install_ihp.sh` script 
- successfully "sees" variables { `PDK_ROOT`, `TOOLS` }, but those come from the `ENV` section of the `base`-stage
- does not "see" variable `OPENVAF_NAME`
   - this is not an `ENV`, but just an `ARG` to the `openvaf`-stage
   - this appears not to be propagated as environmental variables to the `open_pdks`-stage

[image]

hpretl, 2024-01-09:
Yes, all the `ENV`and the `ARG` are available when building with Docker. I know that this is not ideal:

```
OPENVAF_VERSION=$(ls "$TOOLS/$OPENVAF_NAME")
```

If you have something more robust, I take it :-)

martinjankoehler, 2024-01-09:
This PR #32 seems to do the trick.

hpretl, 2024-01-09:
Merged.

PR #32 (2024-01-09): Dockerfile: declare explicit ENV for OPENVAF_NAME
Fixes #31
closed 2024-01-09 by hpretl