iic-jku/IIC-OSIC-TOOLS #33: [Image Build Phase] Modularize dependencies.sh for faster builds

martinjankoehler, 2024-01-09:
**Is your feature request related to a problem? Please describe.**

During stage `base`, the script `_build/images/base/scripts/dependencies.sh` gets executed with a single `RUN` statement.
This means that if the script fails at some point, the whole script has to be re-executed without any partial result caching.

In the current state, the script is quite long running, containing tasks like:
1. Basic APT setup
2. Adding Firefox PPA
3. Preparations for VSCode (add APT repo)
4. Install additional APT Packages (add APT repo)
5. APT Update/Upgrade
6. Git Proxy Setup
7. Download/build/install Boost from source
8. Download/build/install OpenROAD
9. Install PIP packages
10. Cleanup to minimize image size

**Describe the solution you'd like**

By modularizing the `dependencies.sh` script into multiple `RUN` invocations, build and debugging times can be drastically sped up.

- Directory `_build/images/base/scripts/dependencies` contains
   1. `apt_base_install.sh`
   2. `apt_add_firefox_repository.sh`
   3. `apt_add_vscode_repository.sh`
   4. `apt_update_upgrade.sh`
   5. `apt_additional_pkg_install.sh`
   6. `git_proxy_setup.sh`
   7. `boost_install.sh`
   8. `openroad_install.sh`
   9. `pip_pkg_install.sh`
   11. `cleanup_image.sh`

MrHighVoltage, 2024-01-09:
I agree on this, we should modularize the base package dependency.sh script.

Altough I think we should still keep a rather small number of run commands in the Dockerfile, or else we just make a lot of quasi-empty layers. More or less should there be a separate "RUN" in the Dockerfile for each long running command of the dependencies.sh, so we can make use of caching efficiently.

My suggestion:
`base_setup.sh` includes the quick stuff before the huge apt install, like repo adding etc. Maybe also install git here and configure the proxy if needed.
`apt_install_upgrade.sh` does the long running apt install. and update/upgrade stuff.
`install_<pkg>.sh` for each base package, I suggest a separate script. Debugging those package builds can otherwise be a huge hassle.
`finalize.sh` contains the rest of the stuff.

hpretl, 2024-01-10:
I merged https://github.com/iic-jku/IIC-OSIC-TOOLS/pull/26 so we could do a try.

Before we start, one question to the Docker experts here: If we do the cleanup step in a different `RUN` step, is the stuff really removed from the final image? We should not have a situation where we add stuff in one layer, remove stuff in another layer, and the image still as the later-removed contents.

For the further builds this is no large issue, as we selectively copy things into the final image/layers, but for the base there should not be things blowing up the download or image size.

2nd, we should name the stages like

`10_base_setup.sh`
`20_apt_install_upgrade.sh`
`3x_install<pkg>.sh`
`40_finalize.sh`

so there is really no confusion about the sequence :-)

MrHighVoltage, 2024-01-10:
> If we do the cleanup step in a different RUN step, is the stuff really removed from the final image?
It is definitively not visible in the final image, but AFAIK if files are created in one run, and deleted in another, they are still in the lower layer, but marked as deleted in the upper layer.
For that, I suggest we distribute the steps from finalize, that cleanup files to the corresponding script where those files are created.

> Is someone already working on this
I can start doing it now, as I'm currently working on repairing some builds that require changes in the base package.

MrHighVoltage, 2024-01-16:
I finished the modularization of the base installs. It is in the next_release branch.

closed 2024-01-16 by MrHighVoltage