iic-jku/IIC-OSIC-TOOLS #37: Error on image start in `next_release`

hpretl, 2024-01-19:
When starting the latest build `next_release`, the following error message is shown:

```
ERROR: 'klayout' has version (but version must be > 'KLayout 0.28.8')
```

Further we try to consistently use `[ERROR]`, `[WARN]` and `[INFO]` throughout the image, this should be adapted.

martinjankoehler, 2024-01-19:
This looks like a follow-up error from the fact that klayout can't be called ( as @MrHighVoltage  mentioned here https://github.com/iic-jku/IIC-OSIC-TOOLS/pull/25#issuecomment-1900292624 ).

martinjankoehler, 2024-01-19:
Strange, yesterday I built and tested the image using bag2_integration commit b7d394a successfully.
This was based on `next_release` commit e2b85ea9edefbab2b3db003e7f69350b5d5bd12f.

How and how often are the integration tests performed?

The way I tested it was:
```bash
cd IIC-OSIC-TOOLS/_build
docker build . --build-arg http_proxy=$http_proxy --build-arg https_proxy=$https_proxy
```

[image]

closed 2024-07-12 by hpretl