iic-jku/IIC-OSIC-TOOLS #41: openlane --smoke-test fails with: No module named 'pya'

uChoh5fe, 2024-02-09:
If I run `openlane --smoke-test`, I get the following error:

`───────── GDSII Stream Out (KLayout) ─────────`
`Traceback (most recent call last):
File "/usr/local/lib/python3.10/dist-packages/openlane/scripts/klayout/stream_out.py", line 49, in <module>
import pya
ModuleNotFoundError: No module named 'pya'
`

If I install `pya` with `pip3 install pya` the error changed to:

`───────── GDSII Stream Out (KLayout) ─────────`
`/usr/local/lib/python3.10/dist-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due 
to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is 
not available.
warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
module 'pya' has no attribute 'Technology'
`

What do I wrong? 

Any help is appreciated.

Thanks for this nice image anyway :-)

Markus

**Environment:**
 - OS: Debian Linux with apptainer (docker image to sandbox)
 - Operating mode: shell
 - Version tag: latest

**Additional context**
I have build  a sandbox with apptainer from the docker image. Btw docker hub overview mentions RockyLinux 8.5. as base.

hpretl, 2024-02-09:
On which image tag are you? Can you please do an `echo $IIC_OSIC_TOOLS_VERSION`? The OpenLane smoketest works on `2024.01` on my end.

hpretl, 2024-02-09:
You should not use `next`, this is expected to break :-) It is strange that it does not work with `2024.01`. On which OS/CPU are you? Which start script are you using?

uChoh5fe, 2024-02-09:
Ok, I will skip `next`. I build a sandbox with `apptainer build --sandbox /tmp/iic-osic-tools docker://hpretl/iic-osic-tools:latest` and entering with `apptainer exec --writable --no-home /tmp/iic-osic-tools.latest bash -i`. Modify the `export PATH=/foss/tools/bin:$PATH` and run `openlane --smoke-test`.

OS/CPU: Linux Debian bookworm/i5-4590

hpretl, 2024-02-09:
OK, I will try to replicate that on a Debian machine. We are not using such a setup usually, we recommend using Docker and our start scripts. Since you are on Linux you like want to use `start_X.sh`. Is there a possibility you test this?

uChoh5fe, 2024-02-09:
With `apptainer` I don't need to start `X`. Windows will open directly, tested with `openroad -gui` or `klayout` (there is no sym link from `/foss/tools/bin` to `klayout` by purpose?)

hpretl, 2024-02-09:
Here is the solution: After you enter `apptainer`, you need to do this:
```
source /headless/.bashrc
```

Since I am not familiar with Apptainer I don't know why this does not happen by default, but when you do this the OpenLane smoketest runs fine:

```
LVS Done.
───────────────────────────────────────────────────────────────────────────────────────── Layout vs. Schematic Error Checker ─────────────────────────────────────────────────────────────────────────────────────────
[13:22:33] Check for LVS errors clear.                                                                                                                                                                   checker.py:76
           Gating variable for step 'Yosys.EQY' set to 'False'- the step will be skipped.                                                                                                            sequential.py:250
           Skipping step 'Equivalence Check'…                                                                                                                                                        sequential.py:258
           Saving final views to '/tmp/tmpjjn_q40dopenlane2/smoke_test_design/runs/RUN_2024-02-09_13-21-37/final'…                                                                                   sequential.py:294
           ⭕  Flow complete.                                                                                                                                                                        sequential.py:300
Classic - Stage 60 - Equivalence Check ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60/60 0:00:55
           Smoke test passed.
```

uChoh5fe, 2024-02-09:
:-) 

`Smoke test passed.`

Yes, the `/headless/.bashrc` solved the problem. Thanks for support and your fast answer. Highly recommend.

Regards,
Markus

closed 2024-02-09 by uChoh5fe