iic-jku/IIC-OSIC-TOOLS #42: Step 39: gdsii-klayout failed

uChoh5fe, 2024-02-15:
Sorry, it is me again :-( Maybe this is also because of the apptainer image, but maybe you find also a nice solution ;-)

The following command failed:
`Executing "{python3 /foss/tools/openlane/2024.01/scripts/klayout/stream_out.py --output /home/user/designs/regfile_2r1w/runs/full_guide/results/signoff/regfile_2r1w.klayout.gds --lyt /foss/pdks/sky130A/libs.tech/klayout/tech/sky130A.lyt --lym /foss/pdks/sky130A/libs.tech/klayout/tech/sky130A.map --lyp /foss/pdks/sky130A/libs.tech/klayout/tech/sky130A.lyp --top regfile_2r1w --with-gds-file /foss/pdks/sky130A/libs.ref/sky130_fd_sc_hd/gds/sky130_fd_sc_hd.gds --with-gds-file /home/user/designs/mem_1r1w/runs/full_guide/results/final/gds/mem_1r1w.gds --input-lef /home/user/designs/regfile_2r1w/runs/full_guide/tmp/merged.nom.lef /home/user/designs/regfile_2r1w/runs/full_guide/results/routing/regfile_2r1w.def |& tee /dev/null /home/user/designs/regfile_2r1w/runs/full_guide/logs/signoff/39-gdsii-klayout.log}"
`

This I get in `39-gdsii-klayout.log`:

```
Usage: exe [OPTIONS] INPUT
Try 'exe --help' for help.

Error: Missing argument 'INPUT'.
```

Sorry if I bother you too much.

hpretl, 2024-02-16:
Hmh, strange. I'll be out of office for a week, I'll take a look afterwards...

hpretl, 2024-07-12:
From `2024.05` onwards we moved to OpenLane2, so this issue on OpenLane1 should not be relevant any longer. If there is an issue in OL2 please let us know.

closed 2024-07-12 by hpretl