iic-jku/IIC-OSIC-TOOLS #43: ghdl not found in .tcl-interpreter during yosys-synth

GeneralWuschel, 2024-03-10:
I'm new to hardware design in general, so please let me know if I am the error.

**Describe the bug**
openlane2 fails when executed in /foss/designs with 
`openlane -f VHDLClassic --design-dir /foss/designs/*design_name*/ /foss/designs/*design_name*/config.json`
Yosys fails, because some step.tcl cannot find ghdl.
`ERROR: TCL interpreter returned an error: invalid command name "ghdl"`
Logs attached.
The tclsh and the normal bash execute ghdl just fine. 
Verilog flows work like a charme, so I know how to work around by manually translating with ghdl.

I don't know tcl at all, but it seems odd that global variables can get forgotten when executed from other folders.
Thats the reason I figured there might be a problem with Paths or env, hence I post here.
Please let me also know, if you think thats a problem with openlane itself so I can post there.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup a container with the start_vnc.bat
2. connect to the container via your webbrowser
3. create a folder inside /foss/designs
4. create a basic config.json (mine [config.json](https://github.com/iic-jku/IIC-OSIC-TOOLS/files/14548271/config.json))
5. Open a Terminal in /foss/designs
6. execute faulty command from above (without the asterix')
`openlane -f VHDLClassic --design-dir /foss/designs/design_name/ /foss/designs/design_name/config.json`

**Expected behavior**
The flow should continue at least.

**Environment:**
 - OS: Windows, Docker Desktop
 - Operating mode: VNC
 - Version tag latest[03.07.2024]

**Additional context**
Error log of the run:
[error.log](https://github.com/iic-jku/IIC-OSIC-TOOLS/files/14548236/error.log)
Log of the 01-yosys-vhdlsynthesis step:
[yosys-vhdlsynthesis.log](https://github.com/iic-jku/IIC-OSIC-TOOLS/files/14548237/yosys-vhdlsynthesis.log)
COMMANDS of this step-folder contains:
`yosys -c /usr/local/lib/python3.10/dist-packages/openlane/scripts/yosys/synthesize.tcl`

hpretl, 2024-07-17:
This seems to be an issue with OpenLane and how it calls Yosys to enable the GHDL module. I have opened an issue with OpenLane for clarification, see https://github.com/efabless/openlane2/issues/514.

hpretl, 2024-07-17:
We need a wrapper for `yosys` to properly load the modules like (see https://github.com/efabless/openlane2/issues/514#issuecomment-2232835506)
```
#!/bin/sh
exec -a "$0" <path/to/yosys-bin> -m ghdl "$@"
```

hpretl, 2024-07-21:
This is the version that has been implemented. @donn
```
#!/bin/bash
if [[ $1 == "-h" ]]; then
    exec -a "$0" "$TOOLS/yosys/bin/yosys" "$@"
else
    exec -a "$0" "$TOOLS/yosys/bin/yosys" -m ghdl -m systemverilog "$@"
fi
```

hpretl, 2024-07-21:
Implemented in 79a3942263b5babccfc01fa6ab550124875c706b

closed 2024-07-23 by hpretl