iic-jku/IIC-OSIC-TOOLS #44: Getting a Permission Denied error while running the openlane flow

Satvik3799, 2024-04-04:
**Describe the bug**
When running the opnelane flow, it creates a bunch of directories and files. The permission to create them is denied.

**To Reproduce**
Steps to reproduce the behavior:
1. run start_x.sh script
2. Go to the openlane directory.
3. Run "make mount"
4. Run "./flow.tcl -design spm -overwrite"

**Expected behavior**
The entire openlane flow should run.

**Environment:**
 - OS: Linux - Ubuntu 22.04
 - Operating mode: X11
 - Version tag: latest

**Additional context**
When running a container with "start_shell.sh", the default user is root. But while running a container with "start_x.sh", the default user is Inside the container is "designer". The openlane flow doesn't run inside the container started by 'start_x.sh' script, but for the other one it runs. I need to use the GUI of klayout, so I need to run the container with "start_x.sh".

hpretl, 2024-04-16:
@Satvik3799 When you use the `IIC-OSIC-TOOLS` container, OpenLane is already right there, you don't need to start a container for OpenLane. So what you need to do is

1. Run `start_x.sh`
2. Copy the `spm` example to `/foss/designs`
3. Run `flow.tcl -design spm -overwrite`

Satvik3799, 2024-04-16:
It's working. But since I am running things as the user "Designer", how do I become root in the new terminal that is opened when I run the start_x.sh script.

hpretl, 2024-04-16:
You can do `CONTAINER_USER=0 CONTAINER_GROUP=0 ./start_x.sh`, then you are root inside the container. Use with care.

Satvik3799, 2024-04-16:
thank you so much!

closed 2024-04-16 by Satvik3799