iic-jku/IIC-OSIC-TOOLS #62: riscv64-linux-gnu with multilib vs riscv-unknown-elf

phsauter, 2024-08-22:
In our scripts we commonly use the riscv64-unknown-elf-* tools, for example [here](https://github.com/pulp-platform/cheshire/blob/c3e6f0d8e88353c265315b0583c0c893d6ebe269/sw/sw.mk#L10).  
I know riscv64-linux-gnu with multilib can do the same thing but it requires additional flags and dealing with both possibilities (and the possibility of riscv64-linux-gnu without multilib) can be very annoying and hard to test/debug.  
Additionally, it is likely something beginners do not know about and could lead to difficult to understand problems.

For this reason I think it makes more sense to use riscv-unknown-elf in the docker image instead of riscv64-linux-gnu with multilib.  
The only argument I can think of against it is if there are scenarios where people compile linux applications inside this docker. Personally, I think this is very unlikely because at that point you are firmly in software development land and you will likely need additional tools anyway.

This change would just require a small change here: https://github.com/iic-jku/IIC-OSIC-TOOLS/blob/79ffaf2afb272048a846df766af7b74ec5fe4acb/_build/images/riscv-gnu-toolchain/scripts/install.sh#L10-L11

It would need to be:
```sh
../configure --prefix="${TOOLS}/$RISCV_GNU_TOOLCHAIN_NAME"
make -j"$(nproc)"
```

hpretl, 2024-08-22:
As you are the main users for this tool we will adapt it according your needs.

hpretl, 2024-08-25:
Implemented with 4d4db1850cdcdca4fad9a8fc5e6ea626eef67484. Will be part of next release.

hpretl, 2024-08-29:
@phsauter I have just pushed an image with tag `next` including this change, please test it if you have time.

hpretl, 2024-09-15:
Released with `2024.09` tag.

closed 2024-09-15 by hpretl