iic-jku/IIC-OSIC-TOOLS #66: Adding regression test for PULP tools

hpretl, 2024-08-29:
In the last edition, a host of new tools for PULP platform support has been added.

Furthermore, a mini regression suite has been added to test some basic functionalities. Thus, a test for the PULP tools should be added.

hpretl, 2024-08-29:
@phsauter Can you please take a look? I would appreciate it if you could contribute a mini-test for your flow.

Tests should be similar to the existing ones in https://github.com/iic-jku/IIC-OSIC-TOOLS/tree/main/_tests

phsauter, 2024-08-29:
Yes, I should be able to derive something from [croc](https://github.com/pulp-platform/croc), I have to reduce the size and do a bit more with the compiler.

Do you prefer one test that goes through the whole flow or multiple smaller tests that start with some pre-determined state and only test 1-2 steps at once?

hpretl, 2024-08-29:
45min is a bit steep, 3-4min would be great to have (currently the whole suits runs in less than 5mins).

Monolithic or broken into steps? I guess debugging could be a bit easier if it is smaller steps.

hpretl, 2024-09-11:
The test `test_riscv64-unknown-elf_tools.sh` is failing on my end with this failure log:
```
Running: riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32  -mcmodel=medany -static -ffast-math -c main.c -o main.o
Running: riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32  -mcmodel=medany -static -ffast-math -c crt0.S -o crt0.o
Running: riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32  -mcmodel=medany -static -ffast-math -nostartfiles -lm -lgcc -Tlink.ld -o main.elf main.o crt0.o
/foss/tools/riscv-gnu-toolchain/lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/bin/ld: main.o: in function `main':
main.c:(.text+0x30): undefined reference to `__mulsi3'
collect2: error: ld returned 1 exit status
[ERROR] 'riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32  -mcmodel=medany -static -ffast-math -nostartfiles -lm -lgcc -Tlink.ld -o main.elf main.o crt0.o' failed
```

@phsauter Any idea what might be going wrong?

hpretl, 2024-09-11:
It looks like it is either not finding or not linking `libgcc.a`.

phsauter, 2024-09-11:
Thats what I get for testing it with singularity with pass-through...  
I think we need to enable multilib so it can find `libgcc` for rv32i.
https://github.com/iic-jku/IIC-OSIC-TOOLS/blob/next_release/_build/images/riscv-gnu-toolchain/scripts/install.sh#L10
```
../configure --prefix="${TOOLS}/$RISCV_GNU_TOOLCHAIN_NAME" --enable-multilib
```

hpretl, 2024-09-12:
Yes, that did the trick! Works fine now.

hpretl, 2024-09-15:
Released with `2024.09` tag.

closed 2024-09-15 by hpretl