iic-jku/IIC-OSIC-TOOLS #69: Annotation and Mos Parameter

chennakeshavadasa, 2024-08-31:
In the previous version we were able to directly print the mosfet parameters like gm, gds etc but now its just showing notype which was not the case in previous versions. Another problem is with root access where you could switch PDKs. But now you cant edit the .bashrc file even if you do the now the docker crashes. I never faced these issues in the previous version. Please do solve it.

Steps to reproduce the behavior:
1. Do any simulation.
2. Try to print the mosfet parameter by giving command in the xterm terminal ex. display @m.xm1.msky130_fd_pr__nfet_01v8
3. You can see that the mosfet parameters are displayed as notype. This was not the case in previous versions

**Expected behavior**
I
Usually in the previous versions We could directly print these and also switch PDKs which is not possible noe

[image]

**Environment:**
 - OS: Windows
 - Operating mode: X11

hpretl, 2024-09-02:
We have two relevant updates in the latest version, tag `2024.08`.

1. We updated ngspice to version 43.
2. We upgraded the PDKs to the latest version. sky130 introduced therein the long-awaited continuous models.

I guess the latter causes the issues. You could try the following to switch back to the previous PDK. Inside the Docker VM, please type in a shell:
```
export PDK_ROOT=/headless
volare enable bdc9412b3e468c102d01b7cf6337be06ec6e9c9a
```

Can you please test this whether it solves your issue?

hpretl, 2024-09-02:
@michaelk99 Could you please take a look and try to replicate this issue?

michaelk99, 2024-09-02:
@chennakeshavadasa could you please be more specific on how to reproduce this error?

I've just tested the latest container (on Manjaro Linux, using this [project](https://github.com/iic-jku/kvosid_336004_ss24)) and I could not reproduce it. Simulation runs fine and annotation of mos device parameters works as before.

I could also edit ~/.bashrc without any issues.

Have you tried resetting your container?

Maybe provide a simple xschem schematic and a step-by-step description on how you start the container/xschem

Other notes regarding your message:
- For switching between pdks, simply use 
`iic-pdk sg13g2`,
`iic-pdk sky130A` or
`iic-pdk gf180mcuC` 

- If you want root access inside the container, please follow the README (Section 4.2.1 and 4.4.2 may help):
set `CONTAINER_USER=0` and `CONTAINER_GROUP=0` prior to starting the container
You could prepare a simple script, which sets these variables and then starts up the container using `start_x.bat`

chennakeshavadasa, 2024-09-02:
> We have two relevant updates in the latest version, tag `2024.08`.
>
> 1. We updated ngspice to version 43.
> 2. We upgraded the PDKs to the latest version. sky130 introduced therein the long-awaited [...]

I switched to old container itself version 2024.05. It works properly in that version.

chennakeshavadasa, 2024-09-02:
> @chennakeshavadasa could you please be more specific on how to reproduce this error?
>
> I've just tested the latest container (on Manjaro Linux, using this [...]

[image]
[image]
[image]
I tried simple inverter transient simulation in the latest version. That's how I saw this problem.

chennakeshavadasa, 2024-09-02:
*Before :* 
[image]
After I change the Bashrc file to switch to gf180mcuC
[image]
[image]
As you can see even after changing it, it open the same xschem under sky130. But If stop the container and run it again it crashes.

michaelk99, 2024-09-02:
@chennakeshavadasa could you please send the schematic as a file? I don't know what is in your "s1" code block.

chennakeshavadasa, 2024-09-02:
s1: .tran 1m 10m 100u
Im really sorry I deleted the latest container yesterday and downloaded the 2024.05 version. So I don't have the schematic file. I checked if it was an error from my side. But I deleted and installed it 3 more times again from start, I encountered the same error. Can you also add a feature where if the mos exceeds its breakdown voltage it shows some error or warning.

michaelk99, 2024-09-02:
Hi @chennakeshavadasa!

I could now reproduce your results. However, ngspice works as expected. If you want to display the device parameters, you have to save them first. This was always the case. 
If you have a look at the ouput of your `display all` command, there are no vectors that contain the device model parameters. Therefore, they don't exist in your current ngspice **plot**.

If you want to use the display command or print the parameters as you would a results **vector**, you have to save them first with e.g. `.save @m.xm1.msky130_fd_pr__nfet_01v8[gm]` (this is required for every parameter you want as a **vector**).

Alternatively, use the show command. 
- `show` print params of all devices
- `show m.xm1.msky130_fd_pr__nfet_01v8` print params of xm1 only
- `show : gm` print gm of all available devices

chennakeshavadasa, 2024-09-02:
Can you share your schematic file

chennakeshavadasa, 2024-09-02:
[image]
How am I able to do it in older version then ????

michaelk99, 2024-09-02:
Excuse me, I can now reproduce your output of `display @m.xm1.msky130_fd_pr__nfet_01v8` using the `2024.07` container. But I'm not sure if we can help you with this issue. @hpretl ?

Since the main change is the ngspice version it is very likely that this is the cause of this bug.
Unfortunately, I could not find any hints in the ngspice43 change log.

Looking into the ngspice34 manual, it does not state any specific behavior for using `display` with arguments that are not a vector
Page 381: 
> display [varname ...]
> Prints a summary of currently defined vectors, **or of the names** specified. The vectors are sorted
by name unless the variable nosort is set. **The information given is the name of the vector, the
length, the type of the vector, and whether it is real or complex data.** Additionally, one vector
is labeled [scale]. When a command such as plot is given without a vs argument, this scale is
used for the X-axis. It is always the first vector in a rawfile, or the first vector defined in a new
plot. If you undefine the scale (i.e, let TIME = []), one of the remaining vectors becomes the
new scale (which one is unpredictable). You may set the scale to another vector of the plot with
the command setscale (13.5.77).

So it is hard to tell how to fix this.
Please open a bug report [here](https://sourceforge.net/p/ngspice/discussion/).

chennakeshavadasa, 2024-09-02:
So it is a bug?

hpretl, 2024-10-29:
We close this issue as it is related to `ngspice` and needs to be addressed there.

StefanSchippers, 2024-12-28:
why aren't you using print?: 

```
ngspice 208 -> print @m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[l] = 3.000000e-07
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[w] = 2.000000e-06
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[m] = 1.000000e+00
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[nf] = 1.000000e+00
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[sa] = 0.000000e+00
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[sb] = 0.000000e+00
...
...
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[gmbs] = 1.644872e-05
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[gm] = 1.153455e-04
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[gds] = 1.804085e-06
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[vdsat] = 8.898804e-02
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[vth] = 6.135211e-01
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[id] = 6.617659e-06
...
...
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[vdseff] = 8.776572e-02
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[cgso] = 5.110275e-16
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[cgdo] = 4.998776e-16
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[cgbo] = 2.757858e-21
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[weff] = 1.948000e-06
@m.x2.xm1.msky130_fd_pr__nfet_01v8_lvt[leff] = 2.759500e-07
ngspice 209 -> 
```