iic-jku/IIC-OSIC-TOOLS #71: Add Docker Support for Spike, RISC-V GNU Toolchain, and riscv-pk

yigitbektasgursoy, 2024-09-13:
I would like to request the addition of Docker support for the following tools:

1. **Spike (RISC-V ISA Simulator)**  
2. **RISC-V GNU Toolchain**  
3. **riscv-pk (Proxy Kernel)**

If this is possible, it would be greatly appreciated. Thank you for considering this request.

hpretl, 2024-09-13:
2. The RISC-V toolchain is WIP and will be part of the next release, see https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/62.

For 1. and 3. can you please provide a link (GitHub, etc)? As long as the tools are reasonably small we can include it, but we have to carefully manage our image size.

yigitbektasgursoy, 2024-09-13:
1. **Spike (RISCV ISA SIMULATOR**): https://github.com/riscv-software-src/riscv-isa-sim
2. **riscv-pk (Proxy Kernel)**: https://github.com/riscv-software-src/riscv-pk

hpretl, 2024-09-13:
I quickly tested the build of

1. **Spike**: works, reasonably small addition, will add.
2. **PK**: build works when adding `--with-arch=rv64gc_zifencei` to configure step (see https://github.com/riscv-software-src/riscv-pk/issues/260). Will add.

Both will be added to next release.

hpretl, 2024-09-15:
Implemented with bce469fe8acb6d3e930343eef4c433b1299358ba and 88157ba38a616db76ad23983fe214595c23de467

hpretl, 2024-09-15:
Released with `2024.09` tag.

yigitbektasgursoy, 2024-09-21:
> Implemented with [bce469f](https://github.com/iic-jku/IIC-OSIC-TOOLS/commit/bce469fe8acb6d3e930343eef4c433b1299358ba) and [88157ba](https://github.com/iic-jku/IIC-OSIC-TOOLS/commit/88157ba38a616db76ad23983fe214595c23de467)

Could you check whether install riscv-pk (Proxy Kernel)? I could not execute it in docker.

hpretl, 2024-09-22:
I am no expert of the RISC-V toolchain, so I can't really say much about this. There is an executable `pk` in `/foss/tools/riscv-gnu-toolchain/riscv64-unknown-elf/bin`, so you say it does not work as expected?

yigitbektasgursoy, 2024-09-25:
I saw that riscv pk is installed. After your reply,  I checked folder and tried it again but it did not work . Maybe there are some errors which I did not notice. After all , I downloaded qemu  (Quick Emulator) to the docker. It solved my problem for now. Thank you for your efforts.

closed 2024-09-25 by yigitbektasgursoy