iic-jku/IIC-OSIC-TOOLS #75: yosys-abc fails in 2024.09

phsauter, 2024-09-26:
**Describe the bug**
A segfault can be created in yosys-abc using a simple sequence of commands, the combination of `&b -d` and `&nf` seem to be the main factors here.  
This same script works with the following systems:
- Container 2024.07 (running on two different hosts)
- Yosys 0.45 compiled with g++11.4.0 on Ubuntu 20.04
- Yosys 0.45 compiled with g++11.2.0 on Alma 8.8
- Yosys 0.45 compiled with g++13.2.0 on Alma 8.8

Yosys (and with that yosys-abc) was always compiled from a clean repo just that was just cloned.

**To Reproduce**
Steps to reproduce the behavior:
[segfault.zip](https://github.com/user-attachments/files/17150810/segfault.zip)
1. Download the zip
2. Extract and step into the folder
3. Start a shell in container 2024.09
4. run `yosys-abc -f abc.script`
5. it will segfault

**Expected behavior**
It should not be segfaulting.

**Environment:**
 - OS: Linux
 - Operating mode: Shell
 - Version tag 2024.09

**Additional context**
I do not understand what might have gone wrong here, I tried recreating it as close as possible but never got a version of ABC that would segfault at this point.  
I think this issue is rather specific and needs many things to ali
gn for it to occur (the script must trigger it, the input design probably needs to be of a certain form etc).

hpretl, 2024-09-27:
This could be a tough one to debug. @MrHighVoltage could you please take a look?

MrHighVoltage, 2024-09-27:
I could reproduce that segfault, on x86 and aarch64, which makes it seem to be a bug.
Some initial thing I found are suggestig, that it could be from running in Docker (wrongfully reported ulimits). I will build yosys-abc with debug symbols enabled as a first step.

phsauter, 2024-09-27:
If you get stuck just update us, I will ask Alan Mishchenko if he has some time to help debug the problem.

Also I don't think it is an inherent docker thing since at least on my side it works in the 2024.07 container, it might be some weird interaction between multiple things.

MrHighVoltage, 2024-09-30:
Rebuilding abc with debug symbols finally gave some insights:
```
Program received signal SIGSEGV, Segmentation fault.
Tim_ManSetCoArrival (p=0x0, iCo=0, Delay=1) at src/misc/tim/timTime.c:118
118         assert( iCo < p->nCos );
```

The backtrace looks like that:

```
#0  Tim_ManSetCoArrival (p=0x0, iCo=0, Delay=1) at src/misc/tim/timTime.c:118
#1  0x0000aaaae69aaf30 in Nf_ManComputeCuts (p=p@entry=0xaaab1bccb730) at src/aig/gia/giaNf.c:997
#2  0x0000aaaae69b3e90 in Nf_ManPerformMappingInt (pGia=pGia@entry=0xaaab1bcbc010, pPars=pPars@entry=0xfffffc1e4668) at src/aig/gia/giaNf.c:2468                                                                  
#3  0x0000aaaae69b5418 in Nf_ManPerformMapping (p=0xaaab1bcbc010, pPars=pPars@entry=0xfffffc1e4668) at src/aig/gia/giaNf.c:2649                                                                                   
#4  0x0000aaaae6005b90 in Abc_CommandAbc9Nf (pAbc=0xaaab1bbd2e00, argc=3, argv=0xaaab1bbd4780) at src/base/abci/abc.c:43036                                                                                       
#5  0x0000aaaae611f1f8 in CmdCommandDispatch (pAbc=pAbc@entry=0xaaab1bbd2e00, pargc=pargc@entry=0xfffffc1e48e8, pargv=pargv@entry=0xfffffc1e48f0) at src/base/cmd/cmdUtils.c:157                                  
#6  0x0000aaaae61183e4 in Cmd_CommandExecute (pAbc=pAbc@entry=0xaaab1bbd2e00, sCommand=sCommand@entry=0xfffffc1e4988 "&nf -D 10000\n") at src/base/cmd/cmdApi.c:210                                               
#7  0x0000aaaae6116028 in CmdCommandSource (pAbc=0xaaab1bbd2e00, argc=<optimized out>, argv=<optimized out>) at src/base/cmd/cmd.c:795                                                                            
#8  0x0000aaaae611f1f8 in CmdCommandDispatch (pAbc=pAbc@entry=0xaaab1bbd2e00, pargc=pargc@entry=0xfffffc1eca38, pargv=pargv@entry=0xfffffc1eca40) at src/base/cmd/cmdUtils.c:157                                  
#9  0x0000aaaae61183e4 in Cmd_CommandExecute (pAbc=pAbc@entry=0xaaab1bbd2e00, sCommand=sCommand@entry=0xaaab1bbd2a10 "source abc.script") at src/base/cmd/cmdApi.c:210                                            
#10 0x0000aaaae61652f4 in Abc_RealMain (argc=<optimized out>, argv=<optimized out>) at src/base/main/mainReal.c:332                                                                                               
#11 0x0000aaaae6162848 in main (argc=<optimized out>, argv=<optimized out>) at src/base/main/main.c:11
```

As I'm really don't know how to work with abc, could you maybe find out, what happens there? Because from that backtrace, I think this looks more like a bug, not related to the container.

MrHighVoltage, 2024-10-01:
I retried the exact version used in the container outside of the Docker container and it still segfaults. With that, I'm very certain, that this is a newly introduced bug in yosys-abc. I'm closing this issue and forward it to the yosys maintainers.

closed 2024-10-01 by MrHighVoltage