iic-jku/IIC-OSIC-TOOLS #80: Using OpenROAD-flow-scripts with IIC-OSIC-TOOLS

simi1505, 2024-11-29:
**Describe the bug**
Since OpenROAD, Yosys, Klayout etc. are installed within the IIC-OSIC-TOOLS, I tried to get the OpenROAD-flow-scripts (ORFS) running by using the already installed tools.

**What I did**
1.) cloned the ORFS Repo to the foss/designs folder
2.) changed the env.sh file to
```
if [[ "$OSTYPE" == "darwin"* ]]; then
  DIR="$(dirname $(perl -e 'use Cwd "abs_path";print abs_path(shift)' "${BASH_SOURCE[0]}"))"
else
  DIR="$(dirname $(readlink -f "${BASH_SOURCE[0]}"))"
fi

if [ -f /opt/rh/rh-python38/enable ]; then
  source /opt/rh/rh-python38/enable
fi

export OPENROAD=/foss/tools/openroad
echo "OpenROAD: ${OPENROAD}"

export OPENROAD_EXE=/foss/tools/openroad/bin/openroad
echo "OpenROAD EXE: ${OPENROAD_EXE}"

export OPENSTA_EXE=/foss/tools/openroad/bin/sta
echo "OpenSTA EXE: ${OPENSTA_EXE}"

export YOSYS_EXE=/foss/tools/yosys/bin/yosys
echo "Yosys EXE : ${YOSYS_EXE}"

export KLAYOUT_DIR=/foss/tools/klayout
echo "KLayout DIR: ${KLAYOUT_DIR}"

export PATH=/foss/tools/openroad/bin:$PATH
export PATH=/foss/tools/yosys/bin:$PATH
export PATH=${DIR}/dependencies/bin:$PATH

if [[ "$OSTYPE" == "darwin"* ]]; then
  export PATH="/Applications/KLayout/klayout.app/Contents/MacOS:$PATH"
  export PATH="$(brew --prefix bison)/bin:$(brew --prefix flex)/bin:$(brew --prefix tcl-tk)/bin:$PATH"
  export CMAKE_PREFIX_PATH="$(brew --prefix or-tools)"
fi

export FLOW_HOME=$DIR/flow
```

3.) selected IHP-SG13G2 gcd project in Makefile (optional)

**First error**
After `source ./env.sh` and `cd flow` and `make`, the first make produced the following error:
```
[INFO DRT-0180] Post processing.
[WARNING GRT-0246] No diode with LEF class CORE ANTENNACELL found.
Error: detail_route.tcl, 63 expected boolean value but got ""
Command exited with non-zero status 1
Elapsed time: 0:08.40[h:]min:sec. CPU time: user 25.24 sys 0.65 (308%). Peak memory: 946496KB.
make[1]: *** [Makefile:769: do-5_2_route] Error 1
make: *** [Makefile:767: results/nangate45/gcd/base/5_2_route.odb] Error 2
```

**Second error**
After searching after issues in the ORFS repo I found: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2473
After setting `set ::env(SKIP_ANTENNA_REPAIR_POST_DRT) 1` in the `detail_route.tcl`, I get the following error:
```

**Third error**

**Additional context**
I made the issue here since this might be IIC-OSIC-TOOLS related. If not, I will open the same issue directly in the ORFS repo.

hpretl, 2024-11-29:
@alpamps since you recently could run ORFS in the VM: Can you please take a look, or better yet, provide your solution?

alpamps, 2024-12-02:
Same issue as here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2461
Looks like some OR versions have this porblem. 

As I tested the iic-osic-tools docker (same env), it works with OR v2.0-14013-gb16bda7e8 and but it fails with OR v2.0-16235-gedf00dff9. The ORFS docker also works. 

Rebuilding the OpenRoad with the latest version could be a potential fix.

simi1505, 2024-12-02:
Can you share your env.sh and Makefile of the working version?
Thanks.

simi1505, 2024-12-03:
This issue of the ORFS Github page (https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2510) probably confirms that the current OpenROAD version within the IIC-OSIC-Tools docker is not compatible with the current ORFS version.

After commenting out
```
set markerdb [[ord::get_db_block] findMarkerCategory DRC]
if {$markerdb != "NULL" && [$markerdb getMarkerCount] > 0} {
  gui::select_marker_category $markerdb
}
```

in `save_images.tcl` the test `make` for the nangate45 runs through without any error but for the ihp-sg13g2 it still fails. I do not know how advisable it is to comment out various script lines.

hpretl, 2024-12-04:
>  I do not know how advisable it is to comment out various script lines anyway.

Probably not. In the next image release I will try to update OpenROAD to a newer version, if OL2 allows.

hpretl, 2024-12-04:
@simi1505 I have just built and pushed a new image with the latest OpenROAD version. OpenLane2 fails at some steps, but maybe it is good to go for ORFS. You might give it a shot.

simi1505, 2024-12-04:
Thanks a lot! The new OpenROAD version (v2.0-17490-g59ec8fed8) runs like a charm for the `nangate45` and the `sky130hd` PDKs without any modifications in the Makefile or any .tcl scripts, just adapt the env.sh as shown above.

However, the ihp-sg13g2 PDK fails with the following error:
```
==========================================================================
finish report_design_area
--------------------------------------------------------------------------
Design area 27653 u^2 98% utilization.
Elapsed time: 0:03.83[h:]min:sec. CPU time: user 2.28 sys 0.65 (76%). Peak memory: 211544KB.
cp ./results/ihp-sg13g2/gcd/base/5_route.sdc ./results/ihp-sg13g2/gcd/base/6_final.sdc
cp /foss/designs/OpenROAD-flow-scripts/flow/platforms/ihp-sg13g2/lef/sg13g2_tech.lef ./objects/ihp-sg13g2/gcd/base/klayout_tech.lef
SC_LEF_RELATIVE_PATH="$\(env('FLOW_HOME')\)/platforms/ihp-sg13g2/lef/sg13g2_stdcell.lef"; \
OTHER_LEFS_RELATIVE_PATHS=$(echo "<lef-files>$(realpath --relative-to=./results/ihp-sg13g2/gcd/base ./objects/ihp-sg13g2/gcd/base/klayout_tech.lef)</lef-files>"); \
sed 's,<lef-files>.*</lef-files>,<lef-files>'"$SC_LEF_RELATIVE_PATH"'</lef-files>'"$OTHER_LEFS_RELATIVE_PATHS"',g' /foss/designs/OpenROAD-flow-scripts/flow/platforms/ihp-sg13g2/sg13g2.lyt > ./objects/ihp-sg13g2/gcd/base/klayout.lyt
sed -i 's,<map-file>.*</map-file>,<map-file>/foss/designs/OpenROAD-flow-scripts/flow/platforms/ihp-sg13g2/sg13g2.map</map-file>,g' ./objects/ihp-sg13g2/gcd/base/klayout.lyt
(env time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' stdbuf -o L /foss/tools/klayout/klayout -zz -rd design_name=gcd \
        -rd in_def=./results/ihp-sg13g2/gcd/base/6_final.def \
        -rd in_files="/foss/designs/OpenROAD-flow-scripts/flow/platforms/ihp-sg13g2/gds/sg13g2_stdcell.gds  " \
        -rd seal_file="" \
        -rd out_file=./results/ihp-sg13g2/gcd/base/6_1_merged.gds \
        -rd tech_file=./objects/ihp-sg13g2/gcd/base/klayout.lyt \
        -rd layer_map= \
        -r /foss/designs/OpenROAD-flow-scripts/flow/util/def2stream.py) 2>&1 | tee ./logs/ihp-sg13g2/gcd/base/6_1_merge.log
[INFO] Reporting cells prior to loading DEF ...
[INFO] Reading DEF ...
[INFO] Clearing cells...
[INFO] Merging GDS/OAS files...
	/foss/designs/OpenROAD-flow-scripts/flow/platforms/ihp-sg13g2/gds/sg13g2_stdcell.gds
[INFO] Copying toplevel cell 'gcd'
[INFO] Checking for missing cell from GDS/OAS...
[ERROR] LEF Cell 'gcd_DEF_FILL' has no matching GDS/OAS cell. Cell will be empty.
[INFO] Checking for orphan cell in the final layout...
[INFO] No orphan cells
[INFO] Writing out GDS/OAS './results/ihp-sg13g2/gcd/base/6_1_merged.gds'
Command exited with non-zero status 1
Elapsed time: 0:00.59[h:]min:sec. CPU time: user 0.33 sys 0.08 (71%). Peak memory: 128196KB.
make[1]: *** [Makefile:892: do-gds-merged] Error 1
make: *** [Makefile:887: results/ihp-sg13g2/gcd/base/6_1_merged.gds] Error 2
```

This is the same error with the old OpenROAD version after commenting the above-stated lines out. Hence, the newer OpenROAD version fixed the above-listed errors.

simi1505, 2024-12-04:
Okey, strange. If `make` without `nuke` or `clean` is run again, a `6_final.gds` is produced and can be displayed. So, now I have a GDS and a netlist of the test make but can I trust them?

A quick search only resulted in one issue in the ORFS repo: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/1288

alpamps, 2024-12-04:
@simi1505 This means there is an empty cell in the final gds file. You can manually delete it. However, using other examples, or your own circuit/verilog code, you won't have this issue anymore.
@hpretl Thanks for the effert. If openlane can't work with the new version, maybe a consider a openroad downgrade to a working version? (OR v2.0-14013-gb16bda7e8)

alpamps, 2024-12-04:
@simi1505 Try using "make nuke && make" everytime you're using OpenRoad. If the flow goes thoroughly, you can trust that the final gds file is correct. for sky130, easily run PEX and simulate the .spice file using ngspice to see how it works. For sg13g2 there is no pex, but there is a workaround for LVS that you can use to get a spice file out of it (w/o parasitics).

simi1505, 2024-12-05:
Hallelujah, the AMS in Xschem works! I wrote a simple 5-Bit Counter in Verilog, ran ORFS with it, converted the synthesized Verilog into xspice and included it in Xschem (Symbol + Pin Assignment). I still get the above-shown error but it seems that it does not really matter.

Here is the analog-mixed signal simulation in Xschem:
[image]

This would be the layout:
[image]

Thank you everyone for your support. I really appreciate it!

simi1505, 2024-12-09:
This sounds like an excellent idea! The error about empty cells is now also gone. Thanks for the hint.

simi1505, 2024-12-09:
By the way, since setting up the analog-digital mixed signal simulation took me quite some time and should be much easier, I have now published a public GitHub repo as a template. There, you will find a step-by-step instruction and a YouTube video in which everything is explained in more detail. This might be a good starting point if you need AMS simulation in the future.

GitHub repo: https://github.com/simi1505/SG13G2_ASIC-Design-Template
YouTube Video: https://www.youtube.com/watch?v=UrUOg9s7gsM&t=461s

If someone gives this Repo a shot, a short feedback on whether everything is working as expected would be much appreciated.

hpretl, 2024-12-11:
Image `2024.12` released with both OpenROAD versions, closing this issue now.

closed 2024-12-11 by hpretl