iic-jku/IIC-OSIC-TOOLS #86: Xyce simulation does not work with IHP

hpretl, 2024-12-29:
Running the stock IHP simulation example, the following pops up:

[image]

Abdulrahman-AL-Sabagh, 2025-01-11:
Hello
Regarding this problem
I found that a xyce plugin called ADMS should be installed in order to fix this error
There is also a necessary instruction that should be added to the XYCE configurations
[image]
the reference for it in the docs https://github.com/Xyce/Xyce/blob/master/INSTALL.md#adding-the-xyceadms-verilog-a-model-compiler

hpretl, 2025-01-11:
Yes, thanks for the hint! I have meanwhile identified and resolved the issue. The next release will have this plugin for Xyce properly installed.

closed 2025-01-12 by hpretl