iic-jku/IIC-OSIC-TOOLS #92: 2025.1: Xschem closes and the Xming server crashes every 10 minutes

MrHighVoltage, 2025-01-13:
**Describe the bug**
As the title says.
Tagging also bug report #91 

**Environment:**
 - OS: Windows
 - Operating mode: X11 with Xming as X-Server
 - Version tag 2025.01

MrHighVoltage, 2025-01-13:
@simi1505 I haven't used Xming personally for the X-Server, so it is a bit hard to debug. To make it easier:

Did you update Xming also or just the container?
Do the windows just close suddenly, like there is a timeout?

simi1505, 2025-01-13:
- I use Windows OS and I have the image and containers in Docker Desktop. 

- I run the image with Xming X Server for Windows from https://sourceforge.net/projects/xming/

- After using Xschem for several minutes it suddenly freezes for some seconds and Xschem and also the Xming Server close. The Xming Server exits completely, so I have to start it again. Not like if I just close the command line and open it again after running the container again.

hpretl, 2025-01-14:
I have reported this to the Xschem maintainer: https://github.com/StefanSchippers/xschem/issues/276

MrHighVoltage, 2025-01-14:
Do we already have some kind of error log? Or is this always lost due to the whole X-Server crashing?

StefanSchippers, 2025-01-14:
@hpretl @MrHighVoltage It's difficult me to reproduce the environment since I don't have windows computers here, however:
Can you roll back to an older xschem build and see if problem happens again?
For example, after cloning the github xschem repo:
```
# clone the xschem repository into xschem-src
https://github.com/StefanSchippers/xschem.git xschem-src
# enter the source tree
cd xschem-src
# List all tags
git tag
# Checkout an older tag
git checkout tags/3.4.4
# Configure for build;  --prefix (installation directory) is optional
./configure [--prefix ...]
# Compile
make
# Install; sudo is needed if using the default prefix in /usr/local or installing 
# in any system directory 
[sudo] make install
```

Running this older xschem will tell us something. If crash happens again then there is likely some issue with the Xserver, or a very long standing bug nobody reported to date.
If the older xschem is stable we can bisect xschem versions and see the commit that caused the problem.
If someone can do this task that would be of great help.
Thank you.

StefanSchippers, 2025-01-14:
Another test you can do is using the 3 checkbuttons in the Options menu in any combinations and test if crash happens again.
 These are various options that change the drawing methods /fix some inconsistencies seen ot buggy servers.

- No XCopyArea drawing model
- Fix for GPUs with broken tiled fill
- Fix broken RDP mouse coordinates

StefanSchippers, 2025-01-14:
Another test you should do is to go in the Xserver configuration and if possible disable Hardware acceleration for it. OpenGL and hardware acceleration is in general the mother of all graphical problems. 
Going deeper try also to disable hardware acceleration in the Virtual machine engine. This also has shown to fix some VM relate crashes in the past.

StefanSchippers, 2025-01-14:
> Do we already have some kind of error log? Or is this always lost due to the whole X-Server crashing?

It is possible to start xschem in debug mode and save all messages to a log file:
`xschem -d 1 - l log`
You can also increase the debug level (-d 2 or -d 3) but I don't recommend to do that in the first place as it produces a shitton of information.

StefanSchippers, 2025-01-14:
> I can only configure one checkbox during the start of the X-Server:

Have you tried disabling it?
Also try to see if there is a similar option do dieable Acceleration on the Virtual machine.

StefanSchippers, 2025-01-14:
> > It is possible to start xschem in debug mode and save all messages to a log file:
> > `xschem -d 1 - l logfile`
> 
> I have tried this now. Where can I find the logfile? It is not located in the folder where I executed Xschem and also not in the installation folder `foss/tools/xschem/...`.

the log file is in the directory where xschem is started. 
If the file is not there may be the  directory is not writeable.

https://github.com/user-attachments/assets/d1b9ff9a-798a-48a0-ad94-fbf5f51e343c

MrHighVoltage, 2025-01-14:
Okay, I setup a Windows-Workstation with the tools now and I can reproduce this error.
It crashes with OpenGL enabled or disabled, that's not it. Further, I can get it to crash pretty quickly by zooming and scrolling around in Xschem, so it looks more like a race-condition and less like some kind of timeout.
Suddenly, everything stops and crashes, including the X-Server (VcXsrv in this case, which surprises me a bit).

The Logfile does not include any errors, I'm afraid Xschem crashes before it can write the last errors.

StefanSchippers, 2025-01-14:
Can you please try to zoom in/out with ctrl-z / Shift-Z ? does it crash too?

MrHighVoltage, 2025-01-14:
I got some log-infos out of the X-Server, first it spams the following line a few hundret times:

`winCreateDIB: CreateDIBSection() failed`

followed by 
```

winBltExposedWindowRegionShadowGDI - BitBlt failed: 0x00000006
winBltExposedWindowRegionShadowGDI - BitBlt failed: 0x00000006
winCreateDIB: CreateDIBSection() failed
```

Then the log file ends.

> Can you please try to zoom in/out with ctrl-z / Shift-Z ? does it crash too?

Yes, it does too. repeatedly clicking fast without waiting for the previous action to finish, seems to do the trick.

StefanSchippers, 2025-01-14:
Also this link contains some information about setting up VcXsrv in windows WSL2, may be it contains some useful info:
https://jack.kawell.us/posts/ros-windows-wsl2/

also some other info I found lurking into the web:
```
 To make VcXsrv work with WSL 2, you need to add the following two lines in your .bashrc file:

export LIBGL_ALWAYS_INDIRECT=1
```
Although I don't think this affects xschem as it does not use 3D acceleration.

MrHighVoltage, 2025-01-14:
On an empty, grid only, or very simple schematic, it doesn't crash, no matter how fast I zoom/move etc...

simi1505, 2025-01-14:
On an empty Xschem window with or without a grid, it does not crash. If I add only a symbol it crashes shortly after scrolling.

StefanSchippers, 2025-01-14:
Ok, thank you.

The next step is to try a build of xschem without `cairo`.
`Cairo` is the library to render nice character fonts in xschem. Without it xschem displays (quite ugly) vector fonts istead, not using the whole cairo system at all.

To do that edit `config.h` in the top source directory of xschem and comment out (do not set it to 0!) this line:
`/* #define HAS_CAIRO 1 */`
Then rebuild and install xschem.

With Cairo:

[image]

Without Cairo:

[image]

MrHighVoltage, 2025-01-14:
okay, 3.4.4, like you initially suggested, also crashes the same way.

StefanSchippers, 2025-01-14:
Thank you, so this is not related to any recent change / addition in xschem since 3.4.4 was tagged in Oct 2023.
```
commit 6305b52d91bf0c73ea0d65ba9797b4f44d5ff100 (tag: 3.4.4)
Author: stefan schippers <stefan.schippers@gmail.com>
Date:   Mon Oct 9 13:53:57 2023 +0200
```

MrHighVoltage, 2025-01-14:
> To do that edit `config.h` in the top source directory of xschem and comment out (do not set it to 0!) this line:
> `/* #define HAS_CAIRO 1 */`
> Then rebuild and install xschem.

Isn't that already disabled?

StefanSchippers, 2025-01-14:
> > To do that edit `config.h` in the top source directory of xschem and comment out (do not set it to 0!) this line:
> > `/* #define HAS_CAIRO 1 */`
> > Then rebuild and install xschem.
> 
> Isn't that already disabled?

If your config.h has this line already commented, then yes, it is disabled (configure did not dind the needed cairo headers and disabled it already). If so it isn't cairo either causing the crash...

StefanSchippers, 2025-01-14:
Also I see lot of posts about VcXsrv crashes, so it should be nice to test the Cygwin (XWin.exe)  Xserver. I used this server 15 years ago on windows 7, and for these pieces of software older is better in general.
https://a.fsdn.com/con/app/proj/xschem/screenshots/xschem_on_windows.png/max/max/1

MrHighVoltage, 2025-01-14:
I think @simi1505 already tried it with Xming. Is that correct?

StefanSchippers, 2025-01-14:
I strongly sugegst to go for the real thing.
https://a.fsdn.com/con/app/proj/xschem/screenshots/xschem_on_windows.png/max/max/1
This provides (among a complete Posix layer on windows) the Cygwin X server (Cygwin/X).
Al these toy projects like Xming / VcXsrv are doing nothing but adding additional bugs.

Also I see that : 
 **WSLg (Windows Subsystem for Linux GUI) is an extension of WSL 2 that allows you to run Linux GUI (Graphical User Interface) applications directly on Windows 10 and 11, without the need for a third-party X Server like VcXsrv and any additional configurations.**
If this is available I should definitely go that way to reduce dependencies.

StefanSchippers, 2025-01-14:
For the time being I see lot of open tickets (Some very old ones) related to the server crash.
[image]

MrHighVoltage, 2025-01-14:
I also tried Cygwin/X now, same crash behavior, but I get more meaningful log outputs and it segfaults after the same messages as VcXsrv:

```
winCreateDIB: CreateDIBSection() failed
winBltExposedWindowRegionShadowGDI - BitBlt failed: 0x00000006
winBltExposedWindowRegionShadowGDI - BitBlt failed: 0x00000006
winCreateDIB: CreateDIBSection() failed
(EE) Fatal signal received in thread 0xa00000010 [0x3108]
(EE) Segmentation fault at address 0x0
(EE) Install gdb to get reasonable backtraces
(EE) /usr/bin/xorg-backtrace failed with returncode 2
(EE) Fatal server error:
(EE) Caught signal 11 (Segmentation fault). Server aborting
(EE) Server terminated with error (1). Closing log file.
winDeinitMultiWindowWM - Noting shutdown in progress
Aborted
```

Anyways, we could not reproduce this error on Linux or MacOS (with XQuartz), only on Windows. The most promising solution right now seems to be WSLg, also because of reduced installation/maintance effort. But we do have rework the start scripts and documentation and we are currently held up with other important tasks.

So for now, we keep it that way. It seems to only be an issue on huge XSchem-Schematics on Windows with X11 mode. If more reports of this error show up, we might revert to the last container release until this error is properly fixed.

Thanks for your help, so far!

StefanSchippers, 2025-01-14:
As far as I can tell both Xservers fail when trying to create a DIB (Device independent Bitmap). This are areas where the Xserver draws into, the area is then handed over for displaying.

Try to find settings in the WSL2 / Xserver about reserved Video memory and if you find anything increase the limit.
**Try also to run a xschem instance, make the xschem window small and load a big schematic.
Does it still crash the server?**

StefanSchippers, 2025-01-14:
Also try to see if the docker container is not enforcing some limits. (although xschem is really a small application)

StefanSchippers, 2025-01-14:
Can you also try to run xschem out of the container? I mean clone the repo, build and install, then execute outside of any container.

StefanSchippers, 2025-01-28:
@hpretl are there updates on this nasty crash? Just curious if there is something I can investigate on my side (although i don't have windows machines to test on) or if the problem was addressed somewhere else.
Thank you

hpretl, 2025-01-31:
No further action on our side meanwhile (busy on other stuff). Could this here 60d180de33181e8463072dc1ece9e39a4a0b62c7 have something to do with it?

StefanSchippers, 2025-01-31:
I don't think the fix above is related to a X server crash, but who knows. 
This was a nasty xschem bug leading to writes in unallocated memory, so anything can happen. But usually should not crash the X server (that is an external process). Anyway please check you don't have one of the indicated ranges of commits affected by the bug in a docker image.

hpretl, 2025-02-07:
@simi1505 I have made a build with the latest Xschem, Docker image tag `next`. Could you please test this release whether the crash you see is still there?

simi1505, 2025-02-07:
Unfortunately, it is still crashing.

StefanSchippers, 2025-02-09:
@simi1505 @hpretl One thing I do in these cases to debug graphic issues is to run a desktop (or better, just a window manager) in a VNC server on a different linux host on the same local network, set the DISPLAY variable to point to the VNC server and run xschem.

in 100% of the cases I have dealt with so far all graphic glitches, garbage on screen, screen redraw issues, crashes related to graphic operations, errors due to off-spec graphic primitives go away (Nvidia, are you hearing me?).

This does not resolve the problem, but hopefully gives some more information.

It is also possible to directly connect to another X server but modern Linux boxes by default do not accept network connections and you probably need to reconfigure and restart the server and this is a super boring task (and also a temporary security hole).

EngGhaith, 2025-02-22:
I had that problem, but after I have uninstalled the Xsrv and its data, then I have downloaded the latest version it works again without any crash!

StefanSchippers, 2025-02-22:
> I had that problem, but after I have uninstalled the Xsrv and its data, then I have downloaded the latest version it works again without any crash!

@MrHighVoltage @simi1505 @hpretl

hpretl, 2025-02-22:
@EngGhaith Interesting find! @simi1505 Could you please try this on your computer as well? If it works, I will add it to the `KNOWN_ISSUES.md`.

ihagg, 2025-02-22:
Hi everyone, I found a wide variety of problems among my students with both VcXsrv and Xming servers on Windows 11. We decided to switch to MobaXterm (Home Edition, portable) and it turned out to be a stable solution for all students. I hope this feedback helps.

hpretl, 2025-02-23:
@ihagg Thanks for the info! @simi1505 Could you please try this suggestion and see whether it works for you?

simi1505, 2025-02-23:
Hi everyone!

I re-installed VcXsrv and also tried MobaXterm, but unfortunately, it is still crashing. With the `DOCKER_TAG=2024.12` Xschem is in general much smoother.

Did you all use `DOCKER_TAG=next`?

Anyway, thanks for the support!

IlirShala1, 2025-02-23:
Hi,

I have tested two versions of VcXsrv. The newest one, Version: 64.1.20.14.0 (website: https://sourceforge.net/projects/vcxsrv/, doesn’t work for me (crash). However, version 64.1.17.2.0 (website: https://vcxsrv.com/) works very well for me. I worked in xschem for more than half hour and there were no crashes. @simi1505 Can you check this as well?

simi1505, 2025-02-23:
Interestingly, version 64.1.17.2.0 (website: https://vcxsrv.com/) works with `DOCKER_TAG=next`!

I will also try the `latest` version now. Thanks!

hpretl, 2025-02-23:
Findings added to `KNOWN_ISSUES.md` with be1d0a709908caab84c5f9cbc11363ed10217036.

MrHighVoltage, 2025-02-25:
As this is still an ongoing issue, I prepared the work to switch to WSLg as recommended by @StefanSchippers . Some tests showed very promising performance and no crashes. Also the setup gets significantly easier, as no third-party X-Server is used any more.

The (currently untested) script is already pushed to `next_release`, I will test it in the evening. If you have time to test it, there is no need to update the container, but the repository with the new script should be downloaded!

simi1505, 2025-02-25:
I downloaded the IIC-OSIC-TOOLS repo, changed `DOCKER_TAG=next` and updated the image in the Docker (3 days old). If I understand correctly, I do not need the X-Server anymore. So, if I start the container in the Docker App without running X-Server, a command line should start right? This is not the case. What else do I need to do? Thanks!

MrHighVoltage, 2025-02-25:
You have to download the "next_release" branch of the Repo, the container itself should be fine for now (although, there will be some updates, so stay on "next" here for now).

I already tested it and pushed some more modifications. A rebuild of the container will be required, but most of the tools should already work out of the box!

MrHighVoltage, 2025-02-25:
So far, everything, that doesn't require GPU acceleration, works fine and is really quick and seemingly stable.

Although there might be some work to do on GPU acceleration, as the approach on WSLg is the forwarding of a "virtual GPU" that is based on the hosts GPU. For that, it basically forwards the commands to the physical GPU. And as of right now, this is not working and requires some work.

MrHighVoltage, 2025-02-28:
@simi1505 perfect, that is great to hear.

At everyone of you, I'm asking to test the newest "next" tag image and the scripts from "next_release" on your platforms.

I used the opportunity and reworked some startup scripts, so now we have:

- Windows using WSLg with X11 and Wayland support, as well as forwarded VGPU + Drivers. This should even support CUDA, if the drivers are installed in the container.
- For Linux I added Wayland support, the forwarded GPUs/DRI have been implemented for a longer time, but due to the fixed setting of "LIBGL_ALWAYS_INDIRECT", it was practically never used. For XWayland clients it seems to be necessary to use DRI, as they don't implement indirect rendering any more.
- For Mac, nothing should have changed. (Sorry @hpretl  :) )

What needs testing:

- [X] KDE Plasma Wayland: Fully works (3D Acceleration with DRI; Firefox & Surfer & GTK & QT apps with Wayland, rest with X11, tested with Radeon and Intel GPUs.
- [X] KDE Plasma X11: Same, everything works as expected.
- [~] Latest Windows 10 with WSL2 & WSLg: Wayland and X11 applications work super fast and reliable. But, there are some hickups with direct rendering on the VGPU  (Firefox WebGL is unstable at best, GDS3D segfaults, at least with a Radeon GPU on the host. If someone has an NVidia GPU and Intel GPU available, please test those things!). Indirect rendering is not available with WSLg, so this is the only way for GPU acceleration.
- [ ] Windows 11 with WSL2 & WSLg: Probably the same as Win10, or not. If someone could give this a shot (@simi1505 ?)
- [ ] MacOS with XQuarts: Nothing should have changed as I wrote before, but this should be verified, too :) (@hpretl )

Thanks everyone for your participation, I will update this comment as soon as new information becomes available.

hpretl, 2025-03-02:
@MrHighVoltage macOS seems to work fine!

simi1505, 2025-03-04:
@MrHighVoltage 
I have tested the current image with the `next` tag and with the `next_release` scripts for the following tools without any problem:
Xschem, ORFS (with some changes), Surfer, GTKWave, and kLayout.
Surfer now runs really smooth. Previously, I had huge problems using Surfer.

MrHighVoltage, 2025-03-05:
Thanks everyone for testing. I'd say this issue is resolved now.

The only thing still a bit problematic is the GPU acceleration on Windows. It works to some extend, but not very reliable. I will add this to known issues. The workaround is to switch to software rendering, which is slow, but basically always works.

closed 2025-03-05 by MrHighVoltage