iic-jku/IIC-OSIC-TOOLS #93: Klayout Sky130 technology install

njcoburn, 2025-01-14:
Is it possible to install klayout with the sky130 technology node already built in? It is great for adding and dimensioning pcells directly in the GUI.
No worries if not - it is easy to install using klayouts Tools > Manage Packages. Also see instructions for that here: https://github.com/efabless/sky130_klayout_pdk
[image]

hpretl, 2025-01-20:
We'll take a look and try to fix it for the next release.

hpretl, 2025-02-07:
In the next release, we will set `KLAYOUT_PATH` accordingly. Implemented with b87e5624cd336320993f6cc204fedc299470d0f7

njcoburn, 2025-02-07:
Awesome. Will close this now. Looking forward to the next release. Really incredible stuff!

closed 2025-02-07 by njcoburn