iic-jku/IIC-OSIC-TOOLS #95: Unable to launch start_x.sh

samueltensingh, 2025-01-26:
I am using Rocky linux 9 installed with docker and the access is through vnc.
I have cloned the iic-osic-tools in my directory.
If I launch ./start_x.sh in my terminal, I get the following message

```
[samuel@sip-compute-1 iic-osic-tools]$ ./start_x.sh 
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
[INFO] Auto detected Linux.
[INFO] /dev/dri detected, forwarding GPU for graphics acceleration.
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
[WARNING] Container iic-osic-tools_xserver_uid_1001 exists.
[HINT] It can also be restarted with "docker start iic-osic-tools_xserver_uid_1001" or removed with "docker rm iic-osic-tools_xserver_uid_1001" if required.

Press "s" to start, and "r" to remove: s
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
iic-osic-tools_xserver_uid_1001
```

I wanted to run Openroad_flow_scripts designs in this iis-osic-tools docker container. Can you help me to run this design?

samueltensingh, 2025-01-29:
I am able to run Openroad_flow_scripts designs in /home/ubuntu in the iis-osic-tools docker container

hpretl, 2025-01-29:
Ok, it looks like it works now for you. We don't have experience with `podman`, we are usually using Docker. @MrHighVoltage do you have additional comments?

hpretl, 2025-02-04:
@samueltensingh Is this issue still relevant, or can we close it?

closed 2025-02-07 by hpretl

naatje80, 2025-04-24:
I was encountering the same issue. Somehow it has something to do with the podman not having sufficient rights to access the Xserver on the local host. For now I could resolve the issue with: `CONTAINER_USER=0 ./start_x.sh`. If I find a proper solution, I could suggest a pull request.

MrHighVoltage, 2025-04-24:
It makes sense that this originates from using podman instead of Docker. Honestly, I think no one of us ever tested using podman.

I quickly skimmed through the internet a bit, it seems that using UID=0 is kind of the easy solution. Maybe if you could provide the container logs from podman, this would help (when running with your real UID, not as root). I expect it to come from non-working X11 authentication.

naatje80, 2025-04-24:
RedHat based distros default switched to podman (although docker is of course still installable).

The cause of the issue was that various bind mounts were still mounted as root. (More information: https://stackoverflow.com/questions/75817076/no-matter-what-i-do-podman-is-mounting-volumes-as-root). For both start_vnc.sh and start_x.sh I had to add the U,rw (instead of rw) option for the bind mount (except for /tmp/.X11-unix in start_x.sh). This resolved the issue for me.