iic-jku/IIC-OSIC-TOOLS #96: make command fails during global route using Openroad_flow_scripts

samueltensingh, 2025-01-29:
I am using iic-osic-tools and copied ORFS flow inside the docker container and tried to run make command and it fails with the below error message for the default gcd design.

==========================================================================
global route report_design_area
Design area 868 u^2 76% utilization.
[INFO FLW-0007] clock core_clock period 0.460000
[INFO FLW-0008] Clock core_clock period 0.490
[INFO FLW-0009] Clock core_clock slack -0.055
****Error: write_ref_sdc.tcl, 40 invalid command name "sta::endpoint_path_count"****
Command exited with non-zero status 1
Elapsed time: 0:02.37[h:]min:sec. CPU time: user 3.21 sys 0.16 (142%). Peak memory: 360444KB.
make[1]: *** [Makefile:756: do-5_1_grt] Error 1
make: *** [Makefile:754: results/nangate45/gcd/base/5_1_grt.odb] Error 2

Expected Behavior
Complete the ORFS flow

Environment
OpenROAD v2.0-16235-gedf00dff9

To Reproduce
Run make command inside the iis-osic-tools docker container

@SiliconPlatformsLab
@mike-attodevices

hpretl, 2025-01-29:
Can you please try this if it solves your issues? (see https://github.com/iic-jku/IIC-OSIC-TOOLS/blob/main/KNOWN_ISSUES.md#the-openroad-flow-scripts-orfs)

> The ORFS required a recent version of `openroad`. Since image tag `2024.12` a recent version is installed alongside the OpenROAD version required by `openlane`. In order to use the ORFS, before calling the `make` script make sure to set the following env vars:

```
export YOSYS_EXE=$TOOLS/yosys/bin/yosys
export OPENROAD_EXE=$TOOLS/openroad-latest/bin/openroad
export OPENSTA_EXE=$TOOLS/openroad-latest/bin/sta
```

samueltensingh, 2025-01-29:
Thanks for your reply. I tried the three commands but still I have the same error.

$TOOLS/openroad-latest/bin/openroad -version 
v2.0-18100-g2fb3cdf40

samueltensingh, 2025-01-30:
Hello @hpretl the default version of openroad in the docker container is v2.0-16235-gedf00dff9 whereas the version in ORFS (openroad-latest) is v2.0-18100-g2fb3cdf40. 

/foss/designs > openroad -version
v2.0-16235-gedf00dff9
/foss/designs/flow > $TOOLS/openroad-latest/bin/openroad -version
v2.0-18100-g2fb3cdf40

I updated the softlink of openroad to openroadlatest and now in the docker container I am able to see the following

/foss/designs > openroad -version
v2.0-18100-g2fb3cdf40

still I see the same error at GRT
**Error: write_ref_sdc.tcl, 40 invalid command name "sta::endpoint_path_count"**

Looks like the version of ORFS and OR is out of sync. Please can you help to fix this so that I have the latest version in ORFS and OR. 

@maliberty please can you also help in this regard on what can be done to update to the latest version in ORFS and OR. Thank you so much for your inputs as well. @vijayank88 please also give your inputs

The idea is to run the ORFS flow inside the iis-osic-tools docker container.

hpretl, 2025-02-04:
Closing this issue, as also https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/2739#issuecomment-2632817426 is closed.

closed 2025-02-04 by hpretl