iic-jku/IIC-OSIC-TOOLS #97: Using host display scaling settings or changing scaling for the container

ahmetenesturan, 2025-02-07:
I use a 13 inch laptop with 2K resolution and windows created using xhost appear very small.

**Describe the solution you'd like**
Ability to change dpi or scaling factor for the docker container.

**Describe alternatives you've considered**
Tried using parameters like `XORG_DPI` in the `start_x.sh` script.

**Additional context**
I am using Fedora 41 with Wayland.

hpretl, 2025-02-07:
@MrHighVoltage Do you have an idea what we could do here?

ahmetenesturan, 2025-02-19:
I also tried `xrandr --output eDP-1 --scale 2x2` and `xrandr --dpi 192` both in the container and on the host machine, which did not work. I didn't have any problems using integer scaling or higher DPI settings on my older X11 systems, the problem might be caused by XWayland, since I use a Wayland session on my Fedora machine.

MrHighVoltage, 2025-02-19:
As you said you use Fedora with Wayland (and I guess Gnome?), I quickly gave this a shot in my Arch Linux with KDE and Wayland, and here the Display-Scaling is perfectly applied to the Windows from the container, including xschem and magic.

As of right now, I don't have a machine with Gnome and Wayland at hand, so I can't really verify your issue.
Reading up on it, it seems like that the Scaling in Gnome isn't as powerful as it is in KDE, but it is getting there. One solutions seems to be to turn on an experimental feature, like described here: https://wiki.archlinux.org/title/HiDPI#Wayland

If you read further down to XWayland, it seems like there is work in progress, but not yet there. Anyways, for now you should get scaled, if blurred, windows using this experimental feature.

Can you please report back if this solved your issue?

ahmetenesturan, 2025-02-20:
I already use the fractional scaling, but setting it to 200% made it work. It looks blurry, so it is worse for me. Hope XWayland fixes it soon. Thanks for your help.

MrHighVoltage, 2025-02-24:
Okay, then I'm closing this for now. Please feel free to reopen, if the issue persists.

closed 2025-02-24 by MrHighVoltage