iic-jku/IIC-OSIC-TOOLS #98: Adding Spicelib, Ax, and Nevergrad python Packages

NooriDan, 2025-02-08:
**Is your feature request related to a problem? Please describe.**
Yes, it is for analog circuit sizing automation research. I've experimented with Bayesian optimization and Evolutionary algorithms for circuit sizing and I've noticed Spicelib is a better package compared to PySpice for automated netlist manipulation because it allows you to parse and edit the netlist file (they support Ngspice, LTspice, Qspice, and XyceSimulator as well). Spicelib's open-source maintainers are very active and responsive as opposed to PySpice which hasnt been updated recently. PySpice does have better interface for running the simulations but their netlist parser and manipulation is not user-friendly in my opinion.

Moreover, [Ax](https://ax.dev/) for Bayesian optimization and [Nevergrad](https://github.com/facebookresearch/nevergrad) for all kinds of Evolutionary Algorithms are very useful tools for automation.

**Describe the solution you'd like**
It'd be great to have these packages pre-installed in the docker image. I tried to do it but it asks for admin password. 
 
**Describe alternatives you've considered**
I've been running my EDA experiments on my laptop but the open-source PDKs and layout automation tools installed in IIC-OSIC-TOOLS is a great addition to my EDA research. 

**Additional context**
I'm currently working on (still under development) [SymXplorer](https://github.com/NooriDan/SymXplorer).

hpretl, 2025-02-09:
`spicelib` is a good idea, I will definitely add it.

About `ax-platform` and `nevergrad` I am not sure (one pulls `torch` which is a sizeable package), but you should be able to easily install it via PIP with the following command:
```
pip install -U --break-system-packages nevergrad ax-platform
```

Does this work for you or is there an error (I just tried and works).

NooriDan, 2025-02-09:
Awesome thanks, Yes it worked for me!

hpretl, 2025-02-10:
I'll close the issue, `spicelib` will be integrated from the next release on.

closed 2025-02-10 by hpretl