iic-jku/IIC-OSIC-TOOLS #100: /foss/designs folder question

alekseil-20, 2025-02-11:
Hi, everyone! I want to apologize for writing my question under "issues" - it is just because I have not found the proper link to questions page. The container runs perfect for me, but while I am progressing I am ending up with having a lot of designs (schematics) in the design folder and I would like to create sub-directories inside it. However, this does not work - ie creating a folder inside /foss/designs and trying to generate a netlist in xschem for a circuit (schematic) from a self-created folder. Is there a way to organize files inside /foss/designs? 
If there is a solution alredy, please refer me to it. Thanks everyone in advance.

hpretl, 2025-02-14:
This is strange because we frequently do this, and should definitely work. I assume this has something to do with the design folder access rights outside your container.

Can you please provide more details to be to assist you? Which OS are you using, which start command of the container?

hpretl, 2025-02-14:
Please look here, maybe this clarifies things: https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/101#issuecomment-2658629048

alekseil-20, 2025-02-15:
Hi, Harald. Thank you for your response. You are absolutely right - creating a folder inside /foss/designs does not prevents a simulation from running. I have tried creating a simple test setup inside this folder and running ngspice - it works. 
Moreover, I have tried to re-simulate the testbench I have previously faced problems with, which includes a symbol of a circuit I had created previously. Without changing anything there and ensuring that both the testbench itself, a symbol of a circuit (.sym file) and a circuit corresponding to the mentioned .sym file (.sch file) are all placed in the same folder - the simulations runs as expected. 
So, to summarize, probably there was some error in syntax or file placement - nothing to do with a container.
Anyway, thanks again for the time you've spent and sorry for bothering you.

MrHighVoltage, 2025-02-19:
Thanks for clarifying, therewith I'm closing this issue now!

closed 2025-02-19 by MrHighVoltage