iic-jku/IIC-OSIC-TOOLS #101: transfer file with iic-osic-tool

BaoTON1701, 2025-02-12:
Hi guys, 

I'm using iic-osic-tool with linux. 

Normally, I start the server using ./start_x.sh ... 

But today i've realized that i don't know how to take the file from this server or transfer it to the local computer. Even i use the find manager of the pop up terminal (the one appear after command "./start_x.sh"). 

Do you know how we are able to transfer the file between them? 

Thanks

alekseil-20, 2025-02-12:
Hi! I usually used to start with ./start_vnc.sh and for me it works as follows: In the linux terminal on host I write "docker ps" to see the list of containers currently running. After I found a container id, I use "docker cp <container_id>:<file_path_inside_container> <final_destination_path>". This command should work both for vnc-viewer case (command that I am using "./start_vnc.sh") and for your case with ./start_x.sh. 
The command described above works for uploading files from container to your host machine. You can easily find a "reversed" command if you want to upload file from host to container.

BaoTON1701, 2025-02-12:
Ah nice, thank you. 

I confirm it works

hpretl, 2025-02-14:
When you start a container either using `start_x.sh` or `start_vnc.sh` the environment variable `DESIGNS` points to the folder which should be mounted inside the container to `/foss/designs`. If this variable is not set, then the default is `$HOME/eda/designs`.

Assuming you want to mount the folder `/home/user1234/eda` into the container, then use either
```
DESIGNS=/home/user1234/eda ./start_x.sh
```
or
```
DESIGNS=/home/user1234/eda ./start_vnc.sh
```

to start the container.

MrHighVoltage, 2025-02-19:
@BaoTON1701 Could you resolve the issue with the suggestion of @hpretl ? Because this mounted folder is the way to go, as the files are safe against deletion of the container, as they are stored on the hosts file system, not inside the container. Therefore we discourage using `docker cp`.

BaoTON1701, 2025-02-19:
> [@BaoTON1701](https://github.com/BaoTON1701) Could you resolve the issue with the suggestion of [@hpretl](https://github.com/hpretl) ? Because this mounted folder is the way to go, as the files are [...]

Hello, 

Sorry I didn't try that way. 

But for me, i always set the directory for DESIGNS before starting ./start_x.sh, but that doesn't seems to have some link to the folder.

closed 2025-02-19 by BaoTON1701