iic-jku/IIC-OSIC-TOOLS #103: Xyce fails to simulate digital circuits in xschem, but works fine for analog circuits

IlirShala1, 2025-02-23:
Hi, I am trying to simulate a digital circuit using the xyce simulator. The digital circuit was written in Verilog and then synthesized befor being imported into xschem. However, the simulation fails with the following error message:

[image]

It must have something to do with the plugin. My first thought was that the launcher in Xschem might be the issue. However, the simulation works perfectly for analog circuits, so that doesn't seem to be the problem. Here is the code inside the launcher:

[image]

The launcher is a copy from the IHP PDK examples.

Is there a known limitation in Xyce regarding digital circuit simulation, or am I missing some requred configurations?

MrHighVoltage, 2025-02-24:
Can you please provide a bit more information on your simulation setup? Are you trying to simulate on transistor-level or gate-level?
Is it mixed signal?
If possible, please share a schematic to verify the behavior. Thanks!

IlirShala1, 2025-02-24:
The design is purely digital at this stage but will later be part of a mixed-signal. The SPICE netlist is synthesized from Verilog at the gate level, using standard cell libraries. I will send you the schematic and the generated SPICE from a Verilog code

[xschem.zip](https://github.com/user-attachments/files/18943074/xschem.zip)

Let me know if that would be helpful. Thanks!!

MrHighVoltage, 2025-02-24:
Okay, as far as I see, the error comes from a problem, that after 4e-9 seconds of simulation time, the solver can't find a solution anymore. Unfortunately, this is not a issue related to the container, so I'm closing this issue here.

If reporting errors, please always provide the full log-file output, thanks!

```
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
--------------------------------------------------------------------------
data:

*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce DEVELOPMENT-202501121419-(Public_Release-7.8.0-326-gbe78559c)-opensource
***** Date: Mon Feb 24 15:30:17 CET 2025

***** Executing netlist /foss/designs/CurrentSteering_DAC/testbenches/simulations/dac_controller_tb_tran.spice

***** Reading and parsing netlist...
Netlist warning in file
 /foss/pdks/ihp-sg13g2/libs.tech/xyce/models/sg13g2_moslv_parm.lib at or near
 line 61
 No model parameter MUNQSO found for model SG13G2_LV_NMOS_PSP of type
 PSP103_VA, parameter ignored.
Netlist warning in file
 /foss/pdks/ihp-sg13g2/libs.tech/xyce/models/sg13g2_moslv_parm.lib at or near
 line 61
 No model parameter SWNQS found for model SG13G2_LV_NMOS_PSP of type
 PSP103_VA, parameter ignored.
Netlist warning in file
 /foss/pdks/ihp-sg13g2/libs.tech/xyce/models/sg13g2_moslv_parm.lib at or near
 line 211
 No model parameter MUNQSO found for model SG13G2_LV_PMOS_PSP of type
 PSP103_VA, parameter ignored.
Netlist warning in file
 /foss/pdks/ihp-sg13g2/libs.tech/xyce/models/sg13g2_moslv_parm.lib at or near
 line 211
 No model parameter SWNQS found for model SG13G2_LV_PMOS_PSP of type
 PSP103_VA, parameter ignored.
***** Setting up topology...

***** Device Count Summary ...
       PSP103_VA level 1 (ADMS PSP103_VA)     1912
       V level 1 (Independent Voltage Source)   16
       -------------------------------------------
       Total Devices                          1928
***** Setting up matrix structure...
***** Number of Unknowns = 6736
***** Initializing...

***** Beginning DC Operating Point Calculation...

***** Beginning Transient Calculation...

***** Percent complete: 1.00035 %
***** Current system time: Mon Feb 24 15:30:36 2025
***** Estimated time to completion: 26 min., 29 sec.

***** Percent complete: 2.00035 %
***** Current system time: Mon Feb 24 15:30:42 2025
***** Estimated time to completion: 18 min., 5 sec.

***** Percent complete: 3.00036 %
***** Current system time: Mon Feb 24 15:30:48 2025
***** Estimated time to completion: 15 min., 7 sec.
[... 46 lines cut ...]
***** End of Xyce(TM) Simulation
*****

Timing summary of 1 processor
                 Stats                   Count       CPU Time              Wall Time
---------------------------------------- ----- --------------------- ---------------------
Xyce                                         1       41.699 (100.0%)       42.450 (100.0%)
  Analysis                                   1       38.710 (92.83%)       38.991 (91.85%)
    Transient                                1       38.710 (92.83%)       38.991 (91.85%)
      Nonlinear Solve                      503       37.063 (88.88%)       37.326 (87.93%)
        Residual                          2114       31.080 (74.53%)       31.279 (73.69%)
        Jacobian                          1575        2.967 ( 7.12%)        2.996 ( 7.06%)
        Linear Solve                      1575        2.430 ( 5.83%)        2.456 ( 5.79%)
      Successful DCOP Steps                  1        0.004 (<0.01%)        0.004 (<0.01%)
      Successful Step                      445        1.481 ( 3.55%)        1.497 ( 3.53%)
      Failed Steps                          57        0.000 (<0.01%)        0.000 (<0.01%)
  Netlist Import                             1        2.710 ( 6.50%)        2.821 ( 6.65%)
    Parse Context                            1        0.011 ( 0.03%)        0.013 ( 0.03%)
    Distribute Devices                       1        2.646 ( 6.34%)        2.733 ( 6.44%)
    Verify Devices                           1        0.000 (<0.01%)        0.000 (<0.01%)
    Instantiate                              1        0.053 ( 0.13%)        0.057 ( 0.14%)
  Late Initialization                        1        0.262 ( 0.63%)        0.300 ( 0.71%)
    Global Indices                           1        0.013 ( 0.03%)        0.013 ( 0.03%)
  Setup Matrix Structure                     1        0.017 ( 0.04%)        0.021 ( 0.05%)
```

closed 2025-02-24 by MrHighVoltage