iic-jku/IIC-OSIC-TOOLS #105: Distrobox Support

jpf91, 2025-02-24:
Hi there,

thanks a lot for maintaining this great project! It is quite annoying to get all the tools of the flows individually, so such an image is much appreciated!

For the "Using a Local X-Server" use case, a lot of the Docker trickery required to forward the X server, to get file permissions right, to get GPU acceleration etc can be handled generically for any container by [distrobox](https://distrobox.it/). When using it on linux with podman installed, it's probably the easiest way to run GUI apps in docker containers. So I'd  prefer to use the distrobox approach and just use the tool installation from the IIC-OSIC-TOOLS container.

It seems like using the devcontainer image might be a perfect fit for this (I assume it requires a similar  set up for env variables, running commands at startup etc). But it seems the image is missing right now? (See https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/104)

I can already successfully create and use the normal image in distrobox:
```bash
distrobox create -i docker.io/hpretl/iic-osic-tools:2025.01 chip
Creating 'chip' using image docker.io/hpretl/iic-osic-tools:2025.01	 [ OK ]
Distrobox 'chip' successfully created.
To enter, run:

distrobox enter chip

distrobox enter chip
Starting container...                   	 [ OK ]
Installing basic packages...            	 [ OK ]
Setting up devpts mounts...             	 [ OK ]
Setting up read-only mounts...          	 [ OK ]
Setting up read-write mounts...         	 [ OK ]
Setting up host's sockets integration...	 [ OK ]
Integrating host's themes, icons, fonts...	 [ OK ]
Setting up distrobox profile...         	 [ OK ]
Setting up sudo...                      	 [ OK ]
Setting up user's group list...         	 [ OK ]

Container Setup Complete!
📦[jpfau@chip ~]$
```

And the I can run the tools just fine:
```
📦[jpfau@chip ~]$ /foss/tools/klayout/klayout
```

However, I can't figure out whether I'm expected to source some file for setup: The tools are not in $PATH, the webserver is obviously not running etc. So I wonder: is there some startup script that can be reused here and that sets up the tool environments?

MrHighVoltage, 2025-02-25:
Hi @jpf91 ,

Yes, there is a script that setups the environment after the startup, the [env.sh](https://github.com/iic-jku/IIC-OSIC-TOOLS/blob/main/_build/images/iic-osic-tools/skel/dockerstartup/scripts/env.sh). As you can see, this script basically adds all the tool paths to the environment, sets a few variables for the container etc...

Is that what you where looking for? Although Distrobox seems interesting, we are currently not planning to further support it, as we want to keep the setup as simple and straight forward as possible.

jpf91, 2025-02-25:
Yes indeed :-) Thanks for pointing my in the right direction. I was wondering where this file is in the final container, and as it's in `/headless` , I didn't see it. ($HOME is set in the Dockerfile, but distrobox overwrites this variable, so I never saw it. Should have tested without distrobox first ;-)

Any chance we can install this into /etc and source it from /etc/bash.bashrc? This should make it work out-of-the-box for distrobox.

For now I think you can just do:
```bash
distrobox create -i docker.io/hpretl/iic-osic-tools:2025.01 chip
distrobox enter chip
source /headless/.bashrc
```

And this seems to set up things properly. I'll do some more testing and if you want I can add a short section to the README. Seems like the image works out of the box, but of course it'd be better if we could get rid of the manual source command.

hpretl, 2025-02-25:
@MrHighVoltage Should we do this? 

> Any chance we can install this script into /foss and source it from /etc/bash.bashrc?

jpf91, 2025-02-26:
@hpretl @MrHighVoltage that'd be great! I'm not sure if it has any effect for the other use cases of the container though, but I think it shouldn't.

So in the meantime I tested whether `/etc/bash.bashrc` actually is executed when using distrobox. Turns out using `/etc/bashrc` does not work, but sourcing the env script from `/etc/profile` or putting a script in `/etc/profile.d` that sources the env file works. As other files in `/etc/profile.d` set up the default prompt, this new script should probably want be sourced last. So it should be named something like `z99_iic_osic_env.sh`.

MrHighVoltage, 2025-02-26:
Ah, I was a bit quick to close the issue, sorry!

Well, the /etc/profile.d approach makes a lot of work, as it simplifies the whole setup and streamlines the setup of the environment. If, as a "side effect", we get Distrobox supported then, it is actually ideal. I will try it myself and report back on that. Thanks for now!

MrHighVoltage, 2025-03-14:
It requires a bit of rework and testing, as our current implementation of the container startup (X11 mode) does not load the files from /etc/profile.
I'm trying to find a solution and let you know when I found one.

hpretl, 2025-08-25:
This is related to https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/164