iic-jku/IIC-OSIC-TOOLS #106: ORFS Crashing in repair_design

jpf91, 2025-02-26:
**Describe the bug**
Hey there,

if I'm running the ORFS as in this test script: `https://github.com/iic-jku/IIC-OSIC-TOOLS/blob/main/_tests/10/test_orfs_sg13g2.sh`, the openroad-latest binary crashes in the `Perform buffer insertion and gate resizing...` step:

```bash
[INFO GRT-0018] Total wirelength: 6652 um
[INFO GRT-0014] Routed nets: 191
Perform buffer insertion and gate resizing...
repair_design -verbose
Signal 11 received
Stack trace:
 0# 0x000056248BE30CF3 in /foss/tools/openroad-latest/bin/openroad
 1# 0x00007FA0A5181330 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbITerm::getAccessPoints() const in /foss/tools/openroad-latest/bin/openroad
 3# odb::dbInst::destroy(odb::dbInst*) in /foss/tools/openroad-latest/bin/openroad
 4# rsz::Resizer::cellWireDelay(sta::LibertyPort*, sta::LibertyPort*, double, std::unique_ptr<sta::dbSta, std::default_delete<sta::dbSta> >&, float&, float&) in /foss/tools/openroad-latest/bin/openroad
 5# rsz::Resizer::findMaxWireLength(sta::LibertyPort*, sta::Corner const*) in /foss/tools/openroad-latest/bin/openroad
 6# rsz::Resizer::findMaxWireLength1() in /foss/tools/openroad-latest/bin/openroad
 7# 0x000056248C005148 in /foss/tools/openroad-latest/bin/openroad
 8# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# 0x00007FA0A953C6BC in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
11# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
12# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /foss/tools/openroad-latest/bin/openroad
13# 0x000056248BE3043F in /foss/tools/openroad-latest/bin/openroad
14# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# main in /foss/tools/openroad-latest/bin/openroad
16# 0x00007FA0A51661CA in /lib/x86_64-linux-gnu/libc.so.6
17# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
18# _start in /foss/tools/openroad-latest/bin/openroad
Command terminated by signal 11
Elapsed time: 0:03.22[h:]min:sec. CPU time: user 9.87 sys 0.19 (311%). Peak memory: 209416KB.
```

**To Reproduce**

* Use `hpretl/iic-osic-tools-devcontainer:2025.01` container
* Run the commands from `https://github.com/iic-jku/IIC-OSIC-TOOLS/blob/main/_tests/10/test_orfs_sg13g2.sh`
* Instead of `make > $RESULT` I used `make` so I can follow the output directly

**Environment:**
 - OS: Host: Linux
 - Operating mode: X11
 - Version tag: 2025.01
 - ORFS commit 86034ce09a61512316525474299a37fc9791f70c

This seems to be a somewhat recent problem with ORFS. When using Commit `759bc3182091858806b53f09a728eaf5ea6216e5` (~2 months old) this does not happen. Maybe openroad-latest needs to be updated?

hpretl, 2025-02-26:
Can you please test with our latest image tag `2025.02`?

hpretl, 2025-02-26:
The issue is that ORFS and OpenROAD are linked somewhat, so when using the latest ORFS it can easily be that OpenROAD is outdated and has a breaking change. We can only maintain monthly releases, so the chance is there that our latest image has a recent enough OpenROAD.

jpf91, 2025-02-26:
Oh, somehow I didn't see the 2025.02 release. Your explanation makes perfect sense.

I now tested 2025.02 and it's working fine with the latest ORFS revision, so I'm closing this issue 👍 

But maybe as a enhancement suggestion: Would it be possible to ship ORFS as part of the container? This way we'd always have a matching version available.

Another option could be to just note the revisions tested by `test_orfs_sg13g2.sh` for every release. But having it shipped in the container would be more comfortable 😇

closed 2025-02-26 by jpf91

hpretl, 2025-02-27:
I have opened an issue for inclusion of ORFS into the image, see here: https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/107