iic-jku/IIC-OSIC-TOOLS #108: Add mold linker and ccache to speedup Verilator

phsauter, 2025-02-27:
The [Mold](https://github.com/rui314/mold) linker is generally faster than ld, lld and gold. 
[Ccache](https://ccache.dev/) is a compiler cache for gcc, clang etc.

Both are interesting for Verilator, which is able to use mold and ccache to speed up its compiling and linking while building the simulation binary.  
This should reduce Verilators 'compile time' making it more user friendly as a simulator.

Both tools are mentioned as optional but recommended for good performance in the [Verilator install guide](https://github.com/verilator/verilator/blob/master/docs/guide/install.rst)

@paulsc96

hpretl, 2025-02-27:
The two packages are small enough to be added, will be part of the next release.

hpretl, 2025-02-27:
Implememented with c6ca81a4850e2ed1474396c3b55347e36df53684.

hpretl, 2025-03-21:
Implemented with `2025.03`.

closed 2025-03-21 by hpretl