iic-jku/IIC-OSIC-TOOLS #111: KLAYOUT LVS NOT WORKING IN DOCKER

Ikram-rs22, 2025-03-08:
I'm using the ihp open pdk and whenever I try to open the lvs option, this error pops up - 

[image]

How to resolve it and check lvs??

hpretl, 2025-03-12:
Can you please specify with a bit more detail what you are doing? Like provide the files or test cases, and which steps you are doing before trying to run the LVS.

mimeiners, 2025-05-09:
Hello,
during design in ihp-sg13g2 I encountered the same error while doing LVS in GUI mode with Klayout.
IHP's macro will open a user interface, which is not possible, since current Klayout ist build without Qt.
A workaround in current docker-image is calling LVS macros from Klayout's Macro IDE (F5).

hpretl, 2025-05-09:
The fix will be part of the next release, likely tag `2025.05`. It is currently gated by a bug of OpenLane.

hpretl, 2025-05-23:
New image is released, please test it. Closing this issue meanwhile.

closed 2025-05-23 by hpretl