iic-jku/IIC-OSIC-TOOLS #114: Change container handling to Docker compose

hpretl, 2025-04-02:
Currently, we provide custom scripts for multi-user VM handling (`eda_server_restart.sh`, `eda_server_start.sh`, `eda_server_stop.sh`).

Potentially, this could be replaced/streamlined by using Docker compose, and we only provide a script to generate a configuration.

hpretl, 2026-10-27:
Closing this. We are keeping the `eda_server_*.sh` scripts instead of moving to Docker Compose.

Since this issue was opened, the scripts have grown beyond start, stop and restart. They now also cover configuration (`eda_server_conf.sh`), backing up user data dirs (`eda_server_bkdatadir.sh`) and removing them (`eda_server_rmdatadir.sh`). They also support rootless Podman, and that support would get harder with Compose, since `podman compose` and `podman-compose` don't behave the same as Docker Compose. Replacing the scripts with a Compose config generator would mean more maintenance for no real gain.

closed 2026-10-27 by hpretl

MrHighVoltage, 2026-10-27:
I agree, additionally there is https://github.com/MrHighVoltage/COWS running as an semi-experimental deployment, which hands the control to the users.