iic-jku/IIC-OSIC-TOOLS #115: Add a way to diable [INFO] messages

soronpo, 2025-04-05:
**Is your feature request related to a problem? Please describe.**
I want to be able to run command cleanly without added informational messages.

**Describe the solution you'd like**
Add a flag to disable messages in the runner scripts.

hpretl, 2025-04-07:
Which `[INFO]` messages do you mean? The one from the startup script? Can you maybe post an example of what you want removed?

hpretl, 2025-04-12:
This is now implemented with 6e88569e6a392099472667b80a3f527d68111fff.

When the environment variable `IIC_OSIC_TOOLS_QUIET` is set, then there are no `[INFO]` messages anymore. Only `[WARNING]` and `[ERROR]` are shown.

So far, this is only implemented for Linux and macOS. IF this should be happening in the Windows script then someone should let us know.

hpretl, 2025-04-12:
@soronpo Let me know if this is what you were asking for.

soronpo, 2025-04-14:
@hpretl Sorry for the delay in response. Yes, this looks to be perfect. 
Although I do intend to run from Windows too, I intend to directly run the docker command, so Windows `.bat` `[INFO]` are no issue for me. The internal docker scripts were the main problem.

hpretl, 2025-04-14:
@soronpo Alright, we will also tackle the `.bat` then. 

@MrHighVoltage could you please take a look as you seem to be more Windows expert than I. :-)

soronpo, 2025-04-14:
> [@soronpo](https://github.com/soronpo) Alright, we will also tackle the `.bat` then.

Just to clarify again. I'm running from Windows, but since I'm not intend to use the `.bat` script, the internal suppression is sufficient for me. I do think it's better to have equivalent experience for all OS that the IIC-OSIC support.

MrHighVoltage, 2025-05-22:
I reopen because I think we need a bit of rework of the solution.
I'm not a huge fan of completely "dev-nulling" all outputs of the docker commands by default in the start scripts. This will make debugging quite a bit harder. 

To keep it consistent, I will implement the silent mode also for the batch scripts.