iic-jku/IIC-OSIC-TOOLS #116: Docker container crashes when zooming in/out in Xschem

chennakeshavadasa, 2025-04-05:
The IIC-OSIC-TOOLS Docker container crashes when zooming in or out in the Xschem window. This issue has been occurring since the last 3 updates.

Steps to reproduce:

1. Open the Xschem window.
2. Create a .sch file and draw a circuit.
3. Try to zoom in or out using the mouse or keyboard.
4. The container crashes, and all unsaved changes are lost.

Note:
This happens consistently and is not limited to just my device , I was able to reproduce it on other devices too.

No fix has been provided so far.

Please look into this.

Thank You.

hpretl, 2025-04-07:
This is a known issue which has been discussed here: https://github.com/iic-jku/IIC-OSIC-TOOLS/issues/92

The solution is to use WSLg on Windows, the start scripts have been adapted with version `2025.05`. Please try to update to the last version of the GitHub content.

hpretl, 2025-04-10:
@chennakeshavadasa Could you please confirm that the solution proposed above solves your issue?

chennakeshavadasa, 2025-04-10:
I was able to make it work, but requires too many steps. Can you pls fix this bug in the next release so that this need not to be done separately. 
Thank you

closed 2025-04-10 by chennakeshavadasa

hpretl, 2025-04-10:
Thanks for closing the issue, and I am glad you could solve it. 

Relating to fixing this on our end: We would if we knew how. We already did extensive debugging, but this is some weird interaction between Xschem and the window manager in WSL. This only happens on Windows, not on Linux nor macOS. So, unless someone finds the root cause (and we can fix it, which is only possible in Xschem and not in WSL) we are stuck.