iic-jku/IIC-OSIC-TOOLS #117: Switch from `volare` to `ciel`

hpretl, 2025-04-10:
It looks like the PDK manager `volare` is superseded by `ciel`. Need to upgrade all spots.

See: https://github.com/fossi-foundation/ciel

hpretl, 2025-04-10:
@donn It looks like `ciel` is replacing `volare`. When would be a good time to switch?

donn, 2025-04-10:
Ciel is indeed replacing Volare. I wouldn't recommend switching just yet because there's no stable release for Ciel on PyPI (and also because LibreLane isn't out and OpenLane requires Volare)

hpretl, 2025-06-26:
We have made the switch to `ciel` with a08fea7c92903ddb8ab6445c57db174943751872.

donn, 2025-06-26:
Neat but wouldn't this break OpenLane?

hpretl, 2025-06-26:
No, so far I don't see a fail. We anyway alias `openlane` to `openlane --manual-pdk`. Smoke tests run fine.

donn, 2025-06-26:
ah okay