iic-jku/IIC-OSIC-TOOLS #118: Black screen in the docker container

sras202, 2025-04-17:
Hello everyone, 

I have been using this Docker container (https://github.com/iic-jku/IIC-OSIC-TOOLS/tree/main) for quite a while (around 2 years from now), and it was working fine. After the recent Efabless shutdown in March, I planned to switch from Skywater130 PDK to IHP PDK. I followed the instructions as defined on this page (https://iic-jku.github.io/analog-circuit-design/analog_circuit_design.html#setting-up-the-design-directory) for using the IHP PDK. After cloning this repository in the container, I got a warning that the Docker image has to be updated for using the transistor libraries defined in the IHP PDK. I updated the Docker image by pulling the changes. When I tried to mount the container using VNC after the image update, I got the black screen as shown in the figure.  I even killed the previous image and created the new Docker image by cloning the repository again, but the problem still exists. 

Does anyone have an idea how this issue can be resolved? My design work is stuck because of this issue. 

Thanks in advance

Shavika Rastogi
PhD Student
International Centre for Neuromorphic Systems
Western Sydney University, Australia
and
Bioomputation Research Group
University of Hertfordshire, UK

[image]

hpretl, 2025-04-17:
@sras202 Can you please provide a bit more detail:

* Are you using Win/Linux/macOS?
* How are you starting the container?

MrHighVoltage, 2025-04-18:
@sras202 Can you please also provide the logs of the container? You can get the logs directly from the Docker Desktop App, or by running `docker logs <container-name>`.

sras202, 2025-04-19:
@MrHighVoltage @hpretl, the container logs are in the attached text file.

sras202, 2025-04-24:
@MrHighVoltage @hpretl Please let me know the possible workaround for this issue.

MrHighVoltage, 2025-04-24:
Sorry for the late reply.

From the log files I can't see an error why your screen stays black. To make some progress here, I have a few suggestions:

- Move your designs folder. The only "external input" to a newly created container comes from the designs folder, everything else is forced by the containers initial state. Move your designs folder (e.g. rename it) so the start script creates a new, definitely empty one. See if the UI works then.
- Try access through the browser. I saw you used TigerVNC as a client, which is of course much more convenient than the browser. But maybe he problem originates from there.
- As you are on Linux, in general I would suggest you to use the X11/Wayland mode directly. This has better performance and integrates well with your existing desktop. Of course, this **only works on a local machine**, not on a server.

sras202, 2025-04-29:
@MrHighVoltage @hpretl  I tried step 1, and renamed the design folder. After starting the container through ./start_vnc.sh, a new design folder is created, but the screen inside the container is still black. 

I also tried step 3 and used the X11 mode through ./start_x.sh, it gives me the following set of warnings : 

[INFO] Auto detected Linux.
Assuming Wayland Socket /run/user/1000/
[WARNING] Wayland socket could not be found. Falling back to X11
[INFO] /dev/dri detected, forwarding GPU for graphics acceleration.
[INFO] Container does not exist, creating iic-osic-tools_xserver_uid_1000 ...
latest: Pulling from hpretl/iic-osic-tools
Digest: sha256:463faabde380877ab77a43734666010acc825b3865bfc7b6ec4ae1e1624b04cf
Status: Image is up to date for hpretl/iic-osic-tools:latest
docker.io/hpretl/iic-osic-tools:latest
0d99711d05e80713ff6df3670caf754d080ec1e4b9ffb9063c38c8cecd14a330

The X11 terminal starts as below, but it is frozen, and I cannot type anything in it. 

[image]

MrHighVoltage, 2025-04-29:
> Are you aware of any workaround for this issue ?

We have never seen this. It looks like something is blocking in the initial scripts while executing, and I don't really know what. Is there for sure no ".designinit" in the designs folder?

Also, could you please again provide the logs from starting in X11 mode? You should be able to get them by executing ```docker logs iic-osic-tools_xserver_uid_`id -u` ```.

sras202, 2025-05-01:
@MrHighVoltage @hpretl There is no ".designinit" in the designs folder. Here are the logs when I tried to start the container in X11 mode : 
[INFO] USER_ID: 1000, GROUP_ID: 1000
[INFO] Final PATH variable: /foss/tools/bin:/foss/tools/sak/:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/foss/tools/klayout:/foss/tools/libman:/foss/tools/osic-multitool:/foss/tools/rftoolkit/bin:/foss/tools/yosys/bin
[INFO] Final PYTHONPATH variable: /usr/lib/python312.zip:/usr/lib/python3.12:/usr/lib/python3.12/lib-dynload:/usr/local/lib/python3.12/dist-packages:/usr/lib/python3/dist-packages:/foss/tools/ngspyce/local/lib/python3.12/dist-packages:/foss/tools/yosys/share/yosys/python3:/foss/tools/klayout/pymod
[INFO] Auto-selected local X11

[INFO] Waiting until one of the sub-processes stops...

(xfce4-terminal:13): xfce4-terminal-WARNING **: 04:06:29.605: Failed to initialize Xfconf: Failed to close file descriptor for child process (Operation not permitted)

(xfce4-terminal:13): xfce4-terminal-WARNING **: 04:06:29.607: Failed to connect to session manager: Failed to connect to the session manager: SESSION_MANAGER environment variable not defined
[INFO] One sub process stopped, exiting...
[INFO] USER_ID: 1000, GROUP_ID: 1000
[INFO] Final PATH variable: /foss/tools/bin:/foss/tools/sak/:/usr/local/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/foss/tools/klayout:/foss/tools/libman:/foss/tools/osic-multitool:/foss/tools/rftoolkit/bin:/foss/tools/yosys/bin
[INFO] Final PYTHONPATH variable: /usr/lib/python312.zip:/usr/lib/python3.12:/usr/lib/python3.12/lib-dynload:/usr/local/lib/python3.12/dist-packages:/usr/lib/python3/dist-packages:/foss/tools/ngspyce/local/lib/python3.12/dist-packages:/foss/tools/yosys/share/yosys/python3:/foss/tools/klayout/pymod
[INFO] Auto-selected local X11

[INFO] Waiting until one of the sub-processes stops...

(xfce4-terminal:13): xfce4-terminal-WARNING **: 04:51:39.938: Failed to initialize Xfconf: Failed to close file descriptor for child process (Operation not permitted)

(xfce4-terminal:13): xfce4-terminal-WARNING **: 04:51:39.940: Failed to connect to session manager: Failed to connect to the session manager: SESSION_MANAGER environment variable not defined
[INFO] One sub process stopped, exiting...