iic-jku/IIC-OSIC-TOOLS #121: Request to add AUCOHL/Fault for DFT

shail-vaidya, 2025-05-13:
**Is your feature request related to a problem? Please describe.**
Missing DFT Tool: Fault

**Describe the solution you'd like**
Request to add github.com/AUCOHL/Fault to IIC-OSIC-TOOLS

hpretl, 2025-05-16:
We have started working on re-adding Fault, but we are stuck mid-process. The issue with Fault is that it required a huge package (Swift) which we want to avoid.

See this branch on the status of Fault integration: https://github.com/iic-jku/IIC-OSIC-TOOLS/pull/56

@donn Do you have bandwidth to work on this and finalize the PR?

hpretl, 2025-05-23:
Fault is in the process of being converted to a PIP package, see https://pypi.org/project/fault-dft.

Once it is available it will be integrated.

donn, 2025-05-23:
0.9.0.dev5 would technically work already. The thing to keep in mind is that it also requires the latest development version of IcarusVerilog to simulate ihp-sg13g2 and- optionally but really recommended- [Quaigh](https://github.com/coloquinte/quaigh).

hpretl, 2025-05-24:
We will include it in the next release, tag `2025.06`.  I need to check into Quaigh, and how to add it. iVerilog and Yosys are always kept up to date in a release.

donn, 2025-05-24:
For the record, I toyed around with Fault and Quaigh in Docker -- Quaigh doesn't require Rust/Cargo to be installed either after the build is complete. You just copy the binary over from the builder image.

donn, 2025-05-24:
https://github.com/AUCOHL/Fault/releases/tag/0.9.0

donn, 2025-05-24:
it's always something- one sec

donn, 2025-05-24:
0.9.3 is out, should fix that. Try that?

hpretl, 2025-05-25:
Installation works now. If I call `fault` this happens:
```
/foss/designs > fault --version
Traceback (most recent call last):
  File "/usr/local/bin/fault", line 8, in <module>
    sys.exit(exec_fault())
             ^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/fault/__main__.py", line 32, in exec_fault
    os.execle(fault_exe, "fault", *sys.argv[1:], prepare_env())
  File "<frozen os>", line 556, in execle
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

If I trace down the `fault.exe` and start it, then it behaves nicely:
```
/usr/local/lib/python3.12/dist-packages/fault > ./fault.exe
Error: Missing expected argument '--cell-model <cell-model>'

OVERVIEW: Open-source EDA's missing DFT Toolchain

USAGE: fault <subcommand>

OPTIONS:
  --version               Show the version.
  -h, --help              Show help information.

SUBCOMMANDS:
  atpg (default)          Generate/import test vectors for a particular circuit and calculate coverage.
  cut                     Cut away D-flipflops, converting them into inputs and outputs. This is a necessary precursor to the ATPG step.
  synth                   Synthesize Verilog designs using Yosys.
  asm                     Assemble test vectors and golden outputs from JSON and Verilog files.
  tap                     Add a JTAG TAP port and controller to a netlist with a scan-chain.
  chain                   Manipulate a netlist to create a scan chain, and resynthesize.

  See 'fault help <subcommand>' for detailed help.
```

hpretl, 2025-05-25:
Added with 1683dc4a15a40a745e7db65e73fbd2c5019de29e

donn, 2025-05-25:
I'm afraid I can't figure out why this is happening so patches welcome…?

hpretl, 2025-05-25:
@donn I tracked the problem to this part:
```python
ivl_path_rel = os.path.join(os.path.dirname(os.path.dirname(iverilog)), "lib", "ivl")
if not os.path.isdir(ivl_path_rel):
   ivl_path_rel = None
ivl_path = env.get("FAULT_IVL_BASE", ivl_path_rel)
env["FAULT_IVL_BASE"] = ivl_path
```
The var `FAULT_IVL_BASE` is set to `None`, and this triggers the crash. I don't know the function of this, can you help what this IVL is? The variable should be set to some empty string or something, but not `None`.

Maybe a root cause: in our `PATH` we point to `/foss/tools/bin`, which has links to all the binaries, likes `../yosys/bin/yosys`. Perhaps this causes the problem?

hpretl, 2025-05-30:
Fixed with c2e4a7f99fa35e65bcca5a886cb4aacbda54309a.