iic-jku/IIC-OSIC-TOOLS #124: iic-pdk with a bad argument crashes the docker

mattvenn, 2025-05-23:
**To Reproduce**

run: iic-pdk sky130 (instead of iic-pdk sky130A)
the terminal window will close immediately and the docker container crashes

**Expected behavior**
print a message - no such pdk, choose from (list)

**Environment:**
- ubuntu 24
- started with start_x
- don't know the version tag - perhaps include how to find this in the bug template. I started testing today, so the release that was current when this issue was opened.

hpretl, 2025-05-23:
Fixed with 6eb28364ee90b61af8b165a4668ebbd91facdb13

hpretl, 2025-05-24:
The script has been improved, and now now crash anymore. It exits gracefully with an error message if a PDK is not existing.

Example:
```
/foss/tools/bin > sak-pdk sky
[ERROR] PDK directory /foss/pdks/sky not found!
[ERROR] No valid standard cell library selected!
```

Note that the script has been re-named to `sak-pdk`, but `iic-pdk` also works (it's an alias).

hpretl, 2025-05-24:
Fixed in re-release of `2025.05`.

closed 2025-05-24 by hpretl