iic-jku/IIC-OSIC-TOOLS #134: Missing spike in 2025.05

Razi-720, 2025-06-05:
The RISC-V ISA simulator (spike) is missing despite being listed in documentation

hpretl, 2025-06-10:
@MrHighVoltage Georg, do you have an idea? Ideally, when building the RISC-V toolchain, both `spike` and `pk` should be built, but it is not happening. If we can get this to work, then there would be no need for a standalone build of both, as they as components of the toolchain.

stineje, 2025-06-26:
> Re-added dedicated build of Spike with b9e1cc3<https://github.com/iic-jku/IIC-OSIC-TOOLS/commit/b9e1cc389ea48fc3cb47bff403b1339e68093c39>, even though theoretically Spike should be built as part of the tool chain. There is no time to further debug this now, so we add the standalone build as a mid-term solution.

Harald,

We have a script that will install the entire RISC-V tool chain at our repo:  https://github.com/openhwgroup/cvw.  This script just downloads everything and installs it with the right APT items.  Should work with various OSes.  Just clone with —recursive and type “sudo bin/wally-tool-chain-install.sh”.  I wasn’t sure if it was useful, so just fyi.  Take care.

All my best,

James

hpretl, 2025-06-26:
@stineje That's a good hint, thanks! I'll take a look at some time, maybe I get an idea what goes wrong in our build.

closed 2025-07-07 by hpretl