iic-jku/IIC-OSIC-TOOLS #135: X11 forwarding in linux with Docker Desktop

amisapta15, 2025-06-05:
**Describe the bug**
X11 forwarding with the `start_x.sh` script does not work in the Docker image built through Docker Desktop. 

**To Reproduce**
Steps to reproduce the behaviour (in Linux Environment):
1. Download and install Docker Desktop and Pull the IIC-OSIC-Tools image from DockerHub
2.  Clone the repo and try executing the `start_x.sh` to build and start the container with X11 forwarding
3. The following error is shown
```
[INFO] Design directory auto-set to /home/--/eda/designs.
[INFO] Auto detected Linux.
[INFO] Assuming Wayland Socket /run/user/1000/.
[WARNING] Wayland socket could not be found. Falling back to X11.
[INFO] /dev/dri detected, forwarding GPU for graphics acceleration.
[INFO] Container does not exist, creating iic-osic-tools_xserver_uid_1000. Please be patient, this can take a few minutes ...
docker: Error response from daemon: error gathering device information while adding custom device "/dev/dri": no such file or directory

Run 'docker run --help' for more information
```
4. The same is noticed when using Wayland as well.

**Expected behaviour**
I expected to see the terminal running on the host's X11 server of the build container.  

The `start_shell.sh`, `start_vnc`, and `start_jupyter` work, though. I just couldn't start any GUI and forward it to the Host.  The `xhost +local:docker` or `xhost +local:*` or extreme `xhost +` doesn't help.

**Environment:**
 - OS: Linux, Pop!_OS 22.04 LTS
 - Operating mode: X11
 - Version tag: Here is my Neofetch
```
--@pop-os 
         /////////////////////            ------------ 
      ///////*767////////////////         OS: Pop!_OS 22.04 LTS x86_64 
    //////7676767676*//////////////       Kernel: 6.12.10-76061203-generic 
   /////76767//7676767//////////////      Uptime: 14 hours, 23 mins 
  /////767676///*76767///////////////     Packages: 3106 (dpkg), 52 (flatpak) 
 ///////767676///76767.///7676*///////    Shell: bash 5.1.16 
/////////767676//76767///767676////////   Resolution: 2560x1440, 3440x1440 
//////////76767676767////76767/////////   DE: GNOME 42.9 
///////////76767676//////7676//////////   WM: Mutter 
////////////,7676,///////767///////////   WM Theme: Pop 
/////////////*7676///////76////////////   Theme: Pop-dark [GTK2/3] 
///////////////7676////////////////////   Icons: Fluent-purple-dark [GTK2/3] 
 ///////////////7676///767////////////    Terminal: gnome-terminal 
  //////////////////////'////////////     CPU: AMD Ryzen Threadripper 3960X (48 
   //////.7676767676767676767,//////      GPU: NVIDIA GeForce RTX 2080 Ti Rev.  
    /////767676767676767676767/////       Memory: 11259MiB / 64162MiB 
      ///////////////////////////
         /////////////////////                                    
             /////////////                                        

```

**Additional context**
Probable Cause: Docker Desktop is installed without sudo privileges in the Docker Desktop install. It prevents the Docker Build from accessing hardware acceleration. This issue is particular to Linux, I think. 
It is also a well-known issue in the community, and the solution given is to remove the option to use the Host's hardware acceleration, which would necessitate modifications in the start_X script.

hpretl, 2025-06-08:
@MrHighVoltage If the non-sudo privileged Docker is the issue, can we check for this in our startup scripts, and optionally turn-off acceleration?

hpretl, 2025-06-09:
@MrHighVoltage I have tested the updated `start_x.sh` script on macOS, it works.

MrHighVoltage, 2025-06-09:
@amisapta15 I did some further investigation. It seems, that since Docker Desktop on Linux is using a VM like on MacOS, there is no easy way to forward the X11 UI to the container. In the end, we have to do it like on MacOS with XQuartz.

Which means, we have to fall back to TCP to connect to the X-Server, which not only prevents the container from using hardware acceleration, but also reduces the general UI performance from near native to quite bad. Additionally, there is a problem with modern Linux desktop environments, like Gnome and KDE. Since they already run on Wayland, X11 is only supported through XWayland (a compatibility layer) which, to my knowledge, does not support TCP any more for X11 connections....

This sounds rather bad and I'm trying to find a solution. But in general, if you see any chance on changing this, I would strongly advise against using Docker Desktop on Linux...

MrHighVoltage, 2025-06-13:
So, first things first, I reworked this, so it now works on Docker Desktop on Linux.

If Docker Desktop is detected (by checking the version string), an instance of socat is started, which opens the socket on TCP port 6000 (which is display :0 over network). This assumes, that most modern desktop environments do not have TCP connections enabled.

Socat is then running until you either force-stop it with CTRL+c, or when the container quits.

I tested this on KDE Plasma, Wayland and the old X11 mode, further testing would be very welcome.

Also mentioning #136, as I will continue implementing this requests on the ieee_chipathon branch.

amisapta15, 2025-06-14:
No sure, where the Warning is originating from?

amisapta15, 2025-06-26:
Well, it still does not work, but it is a broader problem. So, I agree that, we can close it. 
Can I request a sentence added somewhere in the repo (in KNOWN_ISSUES.md?) for this particular use case Docker Desktop+LINUX+x11-FORWARDING

MrHighVoltage, 2025-06-26:
This is fixed, but your @amisapta15 Linux setup is missing socat installed (That's why the message is: "./start_x.sh: line 119: socat: command not found"). On Ubuntu this should be fixed by just installing socat with apt.

amisapta15, 2025-06-26:
That would have been and probably still is useful. Thanks @MrHighVoltage 

So, Should I make a PR or not, adding socat instalation instruction for that setup?

MrHighVoltage, 2025-06-26:
I implemented a more robust way of handling socat (a message when it is not installed): Fixed in: ed15b9c4b5672540b11a20c9a8d166830ccda579

closed 2025-06-26 by MrHighVoltage

MrHighVoltage, 2025-06-27:
@amisapta15 If you have it ready, please go for it, thanks!

Otherwise, I can quickly add it, too.

amisapta15, 2025-06-27:
Can I request for you quickly add it please?

MrHighVoltage, 2025-06-27:
Added with 1bdbec4c3c001a04c6992493cf00b2fb30cc47a2

MrHighVoltage, 2026-02-08:
Hey!
Let's check a few things quickly:
- Are you sure your repo is up to date? One problem could be, that the script is from an older repository state, without the correct implementation which uses socat.
- You do not have Docker CE and Docker Desktop installed? Check by running `docker version` and see if it reports Desktop or CE.

I can't really think of another problem in the quick, since the start script does more or less exactly this, it verifies which version you are running (CE or Desktop) and deals with it accordingly.

MrHighVoltage, 2026-02-08:
Hi @amisapta15,
the xauthority file is not used in this case with Desktop, since socat is opening the socket as root or the correct user and therefore no xauth is required.

That is why I'm expecting old versions of the start script....