iic-jku/IIC-OSIC-TOOLS #136: Request to add cross-combined build-script for vnc/X11 + Jupyter Server (and adding custom start-up script)

amisapta15, 2025-06-09:
**Is your feature request related to a problem? Please describe.**
Current Docker build scripts support either running VNC or X11 or Shell, or Jupyter Server and use custom names and ports for these various use-cases. While this is great for interacting with a single application seamlessly, but inhibits a multi-application scenario. 

For example,  if I want to run a notebook/code in a Jupyter window and at the same time stream the output of the same to Klayout. This can be done through the X11 script, but it only opens a single port, inhibiting tunnelling of both the X11 and Jupyter at the same time. Same for VNC+Jupyter. 

In addition, being able to execute a custom start-up script at build time would help too. 

**Describe the solution you'd like.**

Two build script pairs
-  `start_vnc+jupyter` (.sh and .bat) + custom startup 
- `start_X11+jupyter` (.sh and .bat) + custom startup   

Where a custom startup  is specifically A Glayout-dev environment setup script, as follows 
[run_GL.sh.txt](https://github.com/user-attachments/files/20654565/run_GL.sh.txt)

**Describe alternatives you've considered**
The alternative is to modify your `start_vnc.sh` or `start_x.sh` to expose an extra port (8888) and execute the startup script attached above after the Docker build process to start Jupyter.  An example is attached with only modification being the addition of port 8888

[start_vnc_GL.sh.txt](https://github.com/user-attachments/files/20654602/start_vnc_GL.sh.txt) 

**Additional context**
This is mainly to set up the gLayout dev-environment in a sandboxed Python, separated from the larger system for the Chipathon Track.  However, I can imagine it being useful in other use cases too.

hpretl, 2025-06-09:
@amisapta15 I discussed with Georg, and we are hesitant to add a few more scripts as we have already quite a few to support, and we anyway plan a unified startscript supporting all different use-cases, but this will take longer.

Proposal for the Chipathon: For using the IIC-OSIC-TOOLS our GitHub repo does not need to be cloned, only the startscripts are needed. This means we could create startscripts specific for the Chipathon, and distribute also via the Chipathon website, e.g. as a Zipfile. In this way getting started would even be simpler, as participants only get a Chipathon-startscript from us, and they need to have Docker installed. That's it.

What do you think?

amisapta15, 2025-06-09:
Dear @hpretl 

I couldn't agree more. Supporting scripts, after writing across OS and version updates, is the hardest. 

I also agree with having a specific `Chipathon-startscript` that can be distributed as a Zip File. I actually forked your repo [here](https://github.com/ReaLLMASIC/IIC-OSIC-TOOLS) (currently it's a replica, not updated yet) in hopes that we make and maintain a gLayout-dev specific script that supports your Docker image. We can put it there and update it for future Chipathons as required. 

I propose that we make a branch out of this submitted issue that you don't push to your main. But I can rebase the fork above on this branch and take over maintenance in the fork. We can keep on syncing between the branch and the fork-main for this and future chipathons.  Your help with the custom startscript is very appreciated.

amisapta15, 2025-06-12:
Hi @MrHighVoltage 
any possible script-update?

MrHighVoltage, 2025-06-16:
Hi @amisapta15 

The start scripts that allow UI and Jupyter in one container are readily available in commit 37ca6c7158edbeaff0c3eb6832953a396c23b9eb.

Further, the start scripts for the jupyter environment to use GLayout are implemented with 050868a7be6cce2da1d47961b90e640ecf450c83.

Note that all scripts for the chipathon have "_chipathon_" in them, to make them easily differentiable.