iic-jku/IIC-OSIC-TOOLS #138: Rework build process into multiple Docker images

hpretl, 2025-06-13:
Currently, the build process creates one large Docker image. The build process is slow, and when debugging build issues, this can become cumbersome. Proposed is to split up the monolithic image into several smaller ones (one per tool and PDK), which are stored on Docker. This would also allow supporting targeted images for different use cases:

- `full` for a full package (all tools and PDKs)
- `analog` for the analog development flow
- `digital` for the digital development flow
- `riscv` for the RISC-V development flow

MrHighVoltage, 2025-07-08:
To streamline with the efforts of #153, the `base` image will be stripped of development packages, only delivering the packages required for the runtime. An additional `base-dev` image will be created, that additionally holds all required build-time dependencies.
Both these images will be built from separate Dockerfiles, allowing the maintainers to specifically choose the time of updating the base images.

hpretl, 2026-10-27:
The build rework is done: per-tool images, separate `base` and `base-dev` images, and one bake DAG in `_build/docker-bake.hcl`. The remaining flavor images (analog, digital, riscv) and the tag scheme moved to #398. Closing this one.

closed 2026-10-27 by hpretl