iic-jku/IIC-OSIC-TOOLS #141: Chipathon 25 Wish: Add a gLayout specific startup script

amisapta15, 2025-06-16:
Opening this issue in response to the mail by Harald. Already, partially being tracked by #136 #135 and can be combined here. 

**Is your feature request related to a problem? Please describe.**

The problem is to set up a Miniconda Python environment isolated from the rest of the system, install a specific Python version, and install a few tools in it, then register that environment with Jupyter at the system level. 

**Describe the solution you'd like**
The goal is to 
- Execute a script on startup 
- Start a Jupyter Server that is forwarded to the Host. 

If you can add a `.designinit` and a `random.sh`, I can create a PR to change the `random.sh`. Since it is sandboxed, it does not interfere with other parts of the system. This script needs to be executed at both build-time and subsequent restart times. The latest version of the script is attached. Will be happy to answer any questions.

[run_GL.sh.txt](https://github.com/user-attachments/files/20752623/run_GL.sh.txt)
 

**Additional context**
Context is covered under #136

MrHighVoltage, 2025-06-16:
Btw.: implemented in 050868a7be6cce2da1d47961b90e640ecf450c83

amisapta15, 2025-06-16:
Hello @MrHighVoltage 

I am afraid, We probably need to keep that option open, as installation of new packages. 

There is a AI/LLM semi-track in Chipathon that allows people to modify the layout in run-time by using ML based optimizations. Not sure if someone will do that or not but, thats a possibility. If that is the case, they probably would want to install pytorch or whatever they want to use. 

So, It will be great if we can keep that option open and keep the environment mutable. 

On same note: two Questions: 

(1) will the start_chipathon_jupyter.sh starts the combined Jupyter+vnc/X11 + excecute the script? 

(2) If needed, How can I update the `run_GL.sh` script?
by, pushing a PR to the ieee_chipathon branch?

amisapta15, 2025-06-16:
(1) Will the `start_chipathon_vnc.sh` or `start_chipathon_jupyter.sh` starts the combined Jupyter + vnc/X11 + excecute the script? 

Because `start_chipathon_vnc.sh` fails with the following (this container doesn't exist)
```bash
[INFO] Design directory auto-set to /home/sapta/eda/designs.
[WARNING] Container iic-osic-tools_xvnc_uid_1000 exists.
[HINT] It can also be restarted with "docker start iic-osic-tools_xvnc_uid_1000" or removed with "docker rm iic-osic-tools_xvnc_uid_1000" if required.
Press "s" to start, and "r" to remove: r
Error response from daemon: No such container: iic-osic-tools_xvnc_uid_1000
```
and the  `start_chipathon_jupyter.sh` successfully runs, but only exposes the port 8888, no VNC or X11. It also doesn't seem to register the GLDEV env in the running Jupyter server. Please see the screenshot attached. 

`python -m ipykernel install --user --name="$ENV_NAME"` This part of the `run_Gl.sh` should have registed the kernel with Jupyter.

[image]

amisapta15, 2025-06-16:
(2) If I want to update `run_GL`, should I push an MR to the `ieee_chipathon` branch, updating the `run_GL.sh` hosted at `_build/images/iic-osic-tools/skel/dockerstartup/scripts`? Or, are you going to integrate the script completely? 

If it's completely integrated, then we can skip the repo cloning part of the script (unnecessarily complicating things, this can be easily done later on the mapped folder!)

```
# === Step 3: Clone repo if needed ===
if [ ! -d "../OpenFASOC" ]; then
    git clone https://github.com/idea-fasoc/OpenFASOC ../OpenFASOC
fi
```

amisapta15, 2025-06-17:
So I attach this slightly modified 'run_GL.sh`?

MrHighVoltage, 2025-06-17:
See [this PR](https://github.com/sscs-ose/sscs-chipathon-2025/pull/16)