iic-jku/IIC-OSIC-TOOLS #145: Add standard cell characterization tools

hpretl, 2025-06-24:
For the characterization of standard cells a few tools exist. `lctime` is already installed in the image, but it would be interesting to also install the following tools and compare them:

* <https://github.com/stineje/CharLib>
* <https://github.com/snishizawa/libretto>

hpretl, 2025-06-26:
Regarding `libretto`: A central installation does not make sense, at it is built in a way that a Makefile (and possibly other things) should be adapted. So do not include into the image.

stineje, 2025-06-26:
pip install git+https://github.com/stineje/charlib

rjordans, 2025-10-16:
After some struggling around I managed to get charlib to run in the docker now!  Thanks for including it.

So, for others trying to figure this out:

The remaining issue I found is that the currently installed version has problems with the ngspice-shared.  If you set `simulator: ngspice-subprocess` in the `settings` block things work nicely again.

rjordans, 2025-10-17:
The downside of running with the ngspice-subprocess is that things seem to be much slower (I guess there might be multi-threading happening in both ngspice and charlib now which overcrowds the CPU)

I found another suggestion that also seems to work: adding the ngspice shared library to the installation using `apt-get install libngspice0-dev`.  Apparently getting it from apt works even with the compiled version being newer than what you'd expect in the apt repository...

https://github.com/stineje/CharLib/issues/71#issuecomment-3140300378

hpretl, 2025-10-17:
The `libngspice.so` is located at `/foss/tools/ngspice/lib`, it is definitely built. We also set the `LD_LIBRARY_PATH=/foss/tools/klayout:/foss/tools/ngspice/lib:/foss/tools/iverilog/lib:/foss/tools/openems/lib:/foss/tools/kactus2` accordingly, so I do not know why `charlib` does not find it.

Maybe @stineje knows why?

rjordans, 2025-10-17:
Looking a bit more into things here is my current trace, the `'ngSpice_Init' not found in library '<None>'` seems to be the problem. 

Looking into `/usr/local/lib/python3.12/dist-packages/PySpice/Spice/NgSpice/Shared.py` I see a bunch of environment variables that are supposed to override this, but PySpice doesn't seem to pick them up if I set them to something that seems reasonable to me (e.g. `export NGSPICE_LIBRARY_PATH=/foss/tools/ngspice/lib/libngspice.so`)

```
> charlib run charlib-inverter.yaml
Searching for YAML files in charlib-inverter.yaml
Reading configuration found in "charlib-inverter.yaml"
Checking configuration syntax
Running input_capacitance for pin A of cell INV
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.12/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/charlib/characterizer/Characterizer.py", line 54, in characterize_cell
    raise e
  File "/usr/local/lib/python3.12/dist-packages/charlib/characterizer/Characterizer.py", line 49, in characterize_cell
    return cell.characterize(self.settings)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/charlib/characterizer/TestManager.py", line 318, in characterize
    input_capacitance = self._run_input_capacitance(settings, pin.name) @ u_F
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/charlib/characterizer/TestManager.py", line 281, in _run_input_capacitance
    simulator = Simulator.factory(simulator=settings.simulator)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/PySpice/Spice/Simulator.py", line 121, in factory
    obj = sub_cls(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/PySpice/Spice/NgSpice/Simulator.py", line 92, in __init__
    self._ngspice_shared = NgSpiceShared.new_instance()
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 457, in new_instance
    instance = cls(ngspice_id=ngspice_id, send_data=send_data, verbose=verbose)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 486, in __init__
    self._init_ngspice(send_data)
  File "/usr/local/lib/python3.12/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 569, in _init_ngspice
    rc = self._ngspice_shared.ngSpice_Init(self._send_char_c,
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/cffi/api.py", line 914, in __getattr__
    make_accessor(name)
  File "/usr/local/lib/python3.12/dist-packages/cffi/api.py", line 910, in make_accessor
    accessors[name](name)
  File "/usr/local/lib/python3.12/dist-packages/cffi/api.py", line 840, in accessor_function
    value = backendlib.load_function(BType, name)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: function/symbol 'ngSpice_Init' not found in library '<None>': /usr/bin/python3: undefined symbol: ngSpice_Init. Did you mean: 'ngSpice_Circ'?
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/bin/charlib", line 7, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/charlib/characterizer/run.py", line 70, in main
    args.func(args)
  File "/usr/local/lib/python3.12/dist-packages/charlib/characterizer/run.py", line 169, in run_charlib
    library = characterizer.characterize()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/charlib/characterizer/Characterizer.py", line 37, in characterize
    cells = pool.map(self.characterize_cell, [*self.tests])
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/multiprocessing/pool.py", line 367, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/multiprocessing/pool.py", line 774, in get
    raise self._value
AttributeError: function/symbol 'ngSpice_Init' not found in library '<None>': /usr/bin/python3: undefined symbol: ngSpice_Init
```

hpretl, 2025-10-17:
@rjordans Could you please provide a testcase so that I can try to see the fail on my setup? Looks like there is an issue with PySpice not looking at the correct spot.

rjordans, 2025-10-17:
charlib-inverter.yaml:
```
settings:
  lib_name:  test_ihp
  units:
    time:               ns
    voltage:            V
    current:            uA
    pulling_resistance: kOhm
    leakage_power:      pW
    capacitive_load:    pF
    energy:             fJ
  named_nodes:
    vdd:
      name: VP
      voltage: 1.2
    vss:
      name: VN
      voltage: 0
  cell_defaults:
    models: 
      - /foss/pdks/ihp-sg13g2/libs.tech/ngspice/models/cornerMOSlv.lib mos_tt
    slews: [0.015, 0.04, 0.08, 0.2, 0.4]
    loads: [0.06, 0.18, 0.42, 0.6, 1.2]

cells:
  inv:
    netlist:    netlist/inv.spice
    area:       128
    inputs:     [A]
    outputs:    ['Y'] # Must be in quotes because YAML interprets Y as boolean True
    functions:  [Y=!A]
```

netlist/inv.spice:
```
.subckt inv VP A Y VN
XM1 Y A VP VP sg13_lv_pmos w=0.15u l=0.13u ng=1 m=1
XM2 Y A VN VN sg13_lv_nmos w=0.15u l=0.13u ng=1 m=1
.ends
```

rjordans, 2025-10-17:
to run, use `charlib run charlib-inverter.spice`

rjordans, 2025-10-17:
OK, let's hope that @stineje has some ideas then.  Otherwise the workaround is able to keep things moving on my side for now

hpretl, 2025-10-17:
It seems to work when I install the `libngspice0` via `apt`. Since the size adder is small, this is probably the simplest fix.

hpretl, 2025-10-17:
Implemented with aed964f41f957445381589f6b49efd39489bd397. Will be part of the next release.

rjordans, 2025-10-17:
(this is maybe not a big problem though as I expect charlib will mainly use the basic ngspice functionality)