iic-jku/IIC-OSIC-TOOLS #153: Drastically reduce image size

hpretl, 2025-07-08:
With all the latest added bells and whistles, the image size is exploding to around 25 GB! This is too large. We need a drastic effort to reduce the image size by implementing the following:

* Separate the `-dev` packages into a dedicated build stage that is not part of the final image.
* Strip binaries and libs wherever possible.
* Use production build settings everywhere (and not debug or development).

hpretl, 2025-07-08:
@MrHighVoltage Here is the issue to track this effort. The target is to have this completed for the `2025.07` release.

MrHighVoltage, 2025-07-21:
For now, the biggest changes we could implement were:

- Getting RISC-V Toolchain more compact with the help of ETH Zürich. (According to them, this were around 400 to 800 MiB). Still, it is the single biggest tool in the container, by far.
- Using the same build-settings for spike, we could save another 500+ MiB.
- Removing java-17 (About 200 to 300 MiB).
- Remove any LLVM-16 dependent libraries (seems there are no system-package dependencies). Currently testing if everything also builds against LLVM-17. Just for openvaf it is necessary to have llvm<=16, but it is linked statically. So the libs are only installed for building the tool.

It is unfortunately complicated to take further steps, since there is a lot of dependencies. The following things are the most-prominent, which we could probably work on:

- Libgraphblas used by libsuitesparse used by Xyce has 175 MiB alone. In general, also the Xyce libs (libxyce.a 160 MiB, libxyce.so nearly 60 MiB ...) and binaries are quite big. Maybe there is potential for optimization.
- There are still 3 versions of libLLVM installed in the system, all of them have system dependencies. So nothing we can really do about it.
- Maybe we can be a bit more stingy on the examples: the analog course pulls 155 MiB, with most of it being the pre-generated files for the sizing (techsweep*.txt and sg13_lv*.mat) and the images for the documentation. Can these files maybe be generated on demand? More or less, if they do no exist, run the generation? Also then they are up-to-date with the newest models anytime.
- Currently, I see there is openroad and openroad-latest. Is this intentionally? (About 120 MiB each)