iic-jku/IIC-OSIC-TOOLS #157: Podman support

MrHighVoltage, 2025-07-14:
Podman is a Container system, similar (and even mostly compatible) to Docker, but has some nice features (like not running as root) and is fully open source.
Like for Docker, there is a Podman Desktop for Linux, Windows and MacOS.

I suggest to support Podman (which can use Docker Images), first just as a platform to run the container (probably simply requires a bit of testing and smaller adaptions to the start script), maybe as a long-term-goal also as a build platform.

Also referencing #105 #95 #45 #31 (most of them are fixed, but the Podman Issues are still unfixed, afaik).

MrHighVoltage, 2025-09-09:
Podman works out of the box as root on Arch Linux. The rootless mode with UID/GID mapping works, although there are problems with X-Forwarding due to limited access permissions (errors because the socket UID/GID does not match the mapped UID/GID).

KoustubhJain, 2026-08-15:
It seems the podman support is still a bit rough around the eges, please see #352

I tried running latest version of IIC-OSIC-Tools via the curl install script on Fedora 44 and it fails to work.